OrcaSlicer/OrcaSlicer #7548: Primer Line option

lincoln12833, 2024-11-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

My printer always oozes when it is heating up. Other slicers have a line which is printed to the side of the model to clear this gunk away before printing

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would like to see a similar option added to orca

MxBrnr, 2024-11-23:
Orca already has this capability, you just have to write in the desired G-code in the `Machine start g-code` box under the Printer Settings. Many printers have this coding written in their firmware so that it does not rely on being entered by the slicer. A quick internet search for "[your printer model] purge line g-code" should net you plenty of results on how to implement this.

lincoln12833, 2024-11-23:
Ok thanks

closed 2025-02-28 by github-actions[bot]