OrcaSlicer/OrcaSlicer #7550: Additional Filament Calibration Tools

jbowers01, 2024-11-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Useful calibration tools are not included such as ironing tests, shrinkage compensation tests, scarf seam tests, small area flow and adaptive pressure advance. such as ironing tests, shrinkage compensation tests, scarf seam tests, small area flow and adaptive pressure advance are not included

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Calibration steps should be added to run calibration tests for these and other beneficial tuning tests. In addition, they should be added as part of the calibration flow for manual tuning under the calibration tab to add the results to the filament profile automatically

### Describe alternatives you've considered

Currently I am opening additional files to perform these calibrations, but not being part of the calibration flow makes it easy to miss steps needed to tweak to perfection,

### Additional context

Orca slicer is the best slicer in my opinion, but I have friends who are afraid of its complexity and don't have an easy path to calibrate their filament profiles. Even experienced users would benefit from these quality-of-life improvements

ls-2037, 2024-11-28:
Adaptive PA calibration would be really useful. Now that the PA pattern calibration prints the acceleration and flow values onto the object, it would be really good to be able to have a set of calibrations automatically generated with appropriate values.

buzzhuzz, 2025-01-09:
@ls-2037 , see https://github.com/SoftFever/OrcaSlicer/pull/7199 for batch mode calibrations.