OrcaSlicer/OrcaSlicer #7555: Reverse on odd does not work when reverse threshhold is set to 0

Mental-Inferno, 2024-11-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Prusa Mini+

### How to reproduce

1. Turn on "Reverse on odd"
2. Set "Reverse threshold" to 0

### Actual results

The perimeters do not alternate extrusion directions (clockwise vs counter-clockwise).

### Expected results

The perimeters should alternate extrusion directions (clockwise vs counter-clockwise).

### Project file & Debug log uploads

[Voron_Design_Cube_v7.zip](https://github.com/user-attachments/files/17892160/Voron_Design_Cube_v7.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

MxBrnr, 2024-11-24:
Update software. This issue has been fixed in OrcaSlicer v2.2.0.

Brendan-Lobo, 2024-11-24:
Just updated, this problem persists

Mental-Inferno, 2024-11-24:
Updated and tired again, this problem persists

Mental-Inferno, 2024-11-25:
Interesting, it seems to be working after I restarted my computer. I didn't know OrcaSlicer requires a computer restart for the update to take effect, I'd assumed that restarting the application would be enough. Closing the issue.

closed 2024-11-25 by Mental-Inferno

MxBrnr, 2024-11-25:
It is weird that is what it needs sometimes; don't know why. Glad we got it worked out for you though.