OrcaSlicer/OrcaSlicer #7562: Pressure Advance Calibration Rejects Manually Set Values > 0.3

rekonnected, 2024-11-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

macOS 15.2 Beta (24C5079e)

### Printer

Bambu Lab A1

### How to reproduce

Go to Calibration tab.
Select 'Manage Result'.
Create or edit a profile.
Set Factor K to any value greater than 0.3.

### Actual results

Per the dialog box that appears, it should accept any value from 0.0 to 1.0.

### Expected results

The box refuses to save any value greater than 0.3.

### Project file & Debug log uploads

[debug_Sun_Nov_24_11_51_09_75643.log](https://github.com/user-attachments/files/17894297/debug_Sun_Nov_24_11_51_09_75643.log)
[nullProject.3mf.zip](https://github.com/user-attachments/files/17894300/nullProject.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Bambu Studio version 1.10.1.50 does not exhibit this issue, and will accept values such as 0.301 and 0.500.
[image]
[image]

[image]

rekonnected, 2025-02-23:
Stale bot-b-gone. Stale bots are not meaningful improvements to issue management and in fact bury issues that are likely still quite valid. 

This issue is still present on the latest release, 2.3.0-beta build dc49116.

Florin-Popescu, 2025-05-09:
Hello! I'm having this issue too. It's annoying with TPU filaments since they teend to have values higher than 0.3 + it's confusing that the dialog says it should go up to 1.0.

rekonnected, 2025-08-16:
Clearly hasn't been resolved yet, so the ever unhelpful stale bot gets to be pushed away again.