OrcaSlicer/OrcaSlicer #7566: Weird artifact on theoretically smooth surface

3nt3, 2024-11-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

n/a

### Printer

Bambu P1P

### How to reproduce

Just slice the project

### Actual results

There are weird artifacts on flat surface and on the top [image]

### Expected results

The layer should be uninterrupted

### Project file & Debug log uploads

[label.zip](https://github.com/user-attachments/files/17895005/label.zip)
[debug_Sun_Nov_24_21_06_59_14728.log.zip](https://github.com/user-attachments/files/17895007/debug_Sun_Nov_24_21_06_59_14728.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MxBrnr, 2024-11-24:
Looks to me like the model is not actually completely flat in those areas, and so the slicer is registering as needing an extra layer there. In my experience, this is most often a result of poor model quality, such as poor model design, or poor resolution when the model was exported from its CAD program.

Noisyfox, 2024-11-27:
Try right click the model and select "Fix Model", which should fix this problem.

closed 2025-03-05 by github-actions[bot]