OrcaSlicer/OrcaSlicer #7567: Set maximum speed in filament

Murolack, 2024-11-25:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have filaments from different vendors and sometimes I even mix them together in one print. But the filaments has different maximum print speeds. I can only set it in the print profile and not in the profile for the filament, so I would so down other filaments, too.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be really helpful if I could set the maximum print speed in the filament profile. So even if I set a higher speed in the print profile the filament maximum will cap it.

VigilantGardener, 2024-11-25:
This protection already exists.

In the filament profile, the filament tab, at the bottom is "Max volumetric speed"(MVS).  This limits the max volume of the filament in mm^3/sec.  If the print speed set in the process would exceed the value set for MVS, the speed will be throttled so that extrusion does not exceed the MVS.

This is a facet of a filament profile that should be tested and set correctly.  There is no one size fits all value for MVS, it will vary by filament type, print temperature, hot end design, nozzle design, maybe other factors too.  For my setup 10mm^3/sec at 210C is great for PLA, but again there are NO one size fits all values.

CNC Kitchen, and others, have tutorials on youtube on how to determine MVS.

At least one one of the filament profiles I had was set to an MVS of 200mm^3/sec, which is ridiculous, and that excessive and unrealistic value did cause problems with prints until it was found and corrected.

VigilantGardener, 2024-11-25:
> But I'm not talking about the mm³/s. I'm talking about the mm/s. As an example I've testet that for my filament a MVS of 18mm³/s with a speed of 170mm/s is perfect. But for other filament I can go up to 250mm/s with even higher MVS. If I set the speed to 250mm/s my "bad" filament is not throttled. I will get bad results. If I'm using only one filament I can set it in the print profile. But if I'm using both together I've so slow down the speed for both.

I think I understand what you're saying, but I do have a question.

Are you using the same filament profile for two different filaments?

Murolack, 2024-11-25:
No, I have two profiles for the filaments. More concrete:
1. filament profile -> Bambu Lab Filament | MVS is set to 21 mm³/s
2. fialment profile -> Redline Filament MVS is set to 18 mm³/s

1. filament can be printed at 250mm/s and 2. only at 170mm/s. 

The other profile I meant is the print profile, where you set you quality, strength, support and the different types of speed. Like first layer speed, outer wall speed and so on.
But this profile is not per filament it is per print.

VigilantGardener, 2024-11-25:
What's the typical nozzle size, line width, and layer height that you print at?

Murolack, 2024-11-25:
Mostly with:
nozzle: 0.4mm
line width: ~0.4mm
layer height: 0.2mm

MxBrnr, 2024-11-25:
Sounds like this is not an enhancement, but rather a bug, where the lower MVS is being ignored. It should be throttling the speed to not exceed the MVS for whichever filament is currently in use.

VigilantGardener, 2024-11-25:
[image]

I think that this is just a matter of settings.  In this image filament 1 has an MVS of 13.6 and filament 2 has an MVS of 20

All that I had to do to accomplish the change in speeds was to set the MVS correctly and then crank the speeds and acceleration until the max speeds were achieved.  To be honest, I just cranked speed and acceleration until the desired results were achieved.  This is something that needs to be really tuned in properly to make sure that the end prints are of good quality.

The formula for MVS based on speed is as follows:
Line Width(mm) x Layer Height(mm) x speed(mm/s)
0.4 x 0.2 x 170 = 13.6
0.4 x 0.2 x 250 = 20.0

Murolack, 2024-11-28:
I think you're totally right. I tried it out and it seems to work. My thought was that these values are completely independent, because if I print with a certain speed and I'm just pushing more filament out it should work. It maybe doesn't make sense, but should work.
So it was my fault and I'm closing this ticket and thank you for clarification.

closed 2024-11-28 by Murolack

VigilantGardener, 2024-11-28:
> I think you're totally right. I tried it out and it seems to work. My thought was that these values are completely independent, because if I print with a certain speed and I'm just pushing more [...]

I'm glad its working for you