OrcaSlicer/OrcaSlicer #7571: Don't add M104 and M140 to start G code if the user defined code contains first_layer_temperature and first_layer_bed_temperature

badVibes--, 2024-11-25:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The slicer checks if user defined Start G-code contains valid M104, M140 commands and adds them if it can not find them. This is a bit awkward when the user defined g-code actually calls a macro on the machine and passes the temperatures to it. 

Example user defined start G code:
 M98 P"/macros/Printer/PrintMacros/PrintStart" T[first_layer_temperature] B[first_layer_bed_temperature]

Currently the slicer still puts additional M104/M140 codes in the output because it can not see they are in the PrintStart macro.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be nice if it also checked for presence of "first_layer_temperature" and "first_layer_bed_temperature" and skipped adding the M104 and M140 if it finds them.

MxBrnr, 2024-11-25:
Looks like this might be exclusive to specific firmware, because it is not happening for me with Klipper, using OrcaSlicer 2.2.0 (stable).
The variables are being passed correctly, and there are no extra M104/140 commands.
[image]
[image]

badVibes--, 2024-11-25:
Might be. I'm using a duet and if my start gcode is:

`M98 P"/macros/Printer/PrintMacros/PrintStart" T[first_layer_temperature] B[first_layer_bed_temperature]`

I get extra temperature control stuff like so:

```
M190 S70 ; set bed temperature and wait for it to be reached
G10 S225 ; set nozzle temperature
;TYPE:Custom
M98 P"/macros/Printer/PrintMacros/PrintStart" T225 B70
```

It's not M104 and M140 like I said though.

If I change my start gcode to:

```
M98 P"/macros/Printer/PrintMacros/PrintStart" T[first_layer_temperature] B[first_layer_bed_temperature]

M104 S[first_layer_temperature]  ; Stops OrcaSlicer from sending temp waits separately
M140 S[first_layer_bed_temperature]
```

then I get:

```
;TYPE:Custom
M98 P"/macros/Printer/PrintMacros/PrintStart" T225 B70

M104 S225  ; Stops OrcaSlicer from sending temp waits separately
M140 S70
```
without the extra temperature stuff above ";TYPE:Custom"

closed 2025-03-03 by github-actions[bot]