OrcaSlicer/OrcaSlicer #7576: Incorrect print time estimates - actual print time 1/3 of the orca estimate

convicte, 2024-11-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows11

### Printer

Anycubic Vyper

### How to reproduce

1. Import model (tested on many models)
2. Slice using standard settings for the printer (tested on different setting combinations)
3. Receive a Total print time 3-4x higher than the actual or one listed for this file by OctoPrint, which both match to within few minutes.

### Actual results

[image]
[image]
[image]
[image]

### Expected results

Print time +/- 5-10% of the actual print time

### Project file & Debug log uploads

[unicorn_pot.zip](https://github.com/user-attachments/files/17909050/unicorn_pot.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

I've printed the same model several times, sliced it several times (to achieve the best result for the use case) and every time the estimated print time was several times the actual time (2h10min vs 7h30min by Orca's estimate).

I recognize there have been numerous attempts at addressing this (#2003, #820, #1969) and they've always died the death of a 'stale' ticket, so naively I am hopeful of a different outcome this time.

At this point in time, Orca is very close to perfect in my day-to-day operation, however this is causing some issues when trying to best allocate printer resources to tasks.

fschlaef, 2024-12-29:
I am facing a similar issue with my Ender 3 Pro running Klipper.
I've found a possible answer in the `Motion ability` tab in the Printer settings.

By default, `Emit limits to G-code` is enabled, although the tooltip specifies that this is ignored for Klipper gcode, so either checked or unchecked makes no difference in the final result.
However, OrcaSlicer still uses the machine limits to compute the estimate time.

Using the default settings below, slicing a standard 20mm Voron cube estimates 33 minutes :

[image]

However, the actual print time was under 22 minutes. I noted that the acceleration was much lower than what my is configured in Klipper, so I matched the X and Y and approximated the E accelaration values as such :

[image]

OrcaSlicer now correctly estimates the same Voron cube in 22 minutes, with no difference to the actual printing because the limits are not emitted to gcode anyway.

Note that SuperSlicer allows for finer control over how machine limits are handled, and the same estimates can be reproduced by changing that value :

[image]

Since Klipper isn't concerned by machine limits gcode, could we also ignore them altogether for time estimates ?

Antwelm, 2025-03-21:
I'm seeing the same on a FlashForge Adventurer 3+. Gives an estimate of 2x the actual print time.

OrcaSlicer 2.3.0-rc

convicte, 2025-06-20:
This remains an issue. No change since the original report.