OrcaSlicer/OrcaSlicer #7578: STEP file import problem

pmainvil, 2024-11-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

10

### Printer

Elegoo Neptune 4 Pro

### How to reproduce

Simply load attached BugStep-Body.step file and render is wrong in the prepare window.    (internal cut sphere rendered as solid)

[image]

If .stl file is loaded render is OK but tesselation reduces quality of print.

[image]

The files were created with FreeCAD 1.0.   They render OK if re-imported in FreeCAD and also on online free 3d viewer https://3dviewer.net/

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[BugStep.zip](https://github.com/user-attachments/files/17909510/BugStep.zip)

Model, .STEP, .STL with Project file.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

pmainvil, 2024-11-25:
[debug_Mon_Nov_25_17_26_04_21468.log.0.zip](https://github.com/user-attachments/files/17909573/debug_Mon_Nov_25_17_26_04_21468.log.0.zip)

Noisyfox, 2024-11-27:
This seems to be a bug in OCCT 7.6.x, which should have been fixed in OCCT 7.7+. However any version newer than 7.6.1 have another critical bug that prevents us from upgrading it: https://github.com/prusa3d/PrusaSlicer/issues/12122 and https://github.com/prusa3d/PrusaSlicer/commit/c6a02106fd1d3caa9a48a6b7c2bdd04546b24485

pmainvil, 2024-12-10:
Additionnal information.  The Step file loads OK in BambuLab 1.10.1.50

[image]

Noisyfox, 2024-12-11:
[image]
Weird... It doesn't work on my side...

pmainvil, 2024-12-11:
I tried again.  My mistake. Please disregard my previous comment about Bambustudio.  

I had loaded the STL by mistake instead of the STEP as confirmed by recent files in Bambu.

Sorry for the trouble.

pmainvil, 2025-03-12:
Problem is still present in V 2.3.0-rc

pmainvil, 2025-06-11:
Still present in 2.3.0 Release

closed 2026-10-04 by RF47