OrcaSlicer/OrcaSlicer #7582: Temperature always be zero in preview for Creality K1 series

adoyle-h, 2024-11-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Creality K1, K1C, K1Max, K1SE 

### How to reproduce

1. Set up Printer
2. Select any Creality K1, K1C, K1 Max Profile
3. Put an stl file with or without changing Printer, Material, Process settings
4. Slice
5. Go to Preview Screen
6. Select Temperature in dropdown menu

### Actual results

In dropdown menu print(nozzle) Temperature shows 0.
Same steps with [Creality Print](https://www.crealitycloud.com/software-firmware/software/creality-print) not give this issue.

### Expected results

The temperature dropdown menu in slicer preview screen must show the set temperature in the material profile.
Printer must use the slicer temperature values.

### Project file & Debug log uploads

[Cube.zip](https://github.com/user-attachments/files/17912866/Cube.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

[image]

Related issue: https://github.com/SoftFever/OrcaSlicer/issues/5491

Sacrasang, 2024-11-27:
I am experiencing the same behavior in version 2.2.0 with a different printer (a modified Ender 5 S1 running Klipper and using the klipper gcode flavor). 

I think the issue is related to this machine's start gcode not including any specific instructions for temperature and is only a graphical error in the gcode reader. 

My machine start gcode is:

start_print BED_TEMP=[bed_temperature_initial_layer_single] EXTRUDER_TEMP=[nozzle_temperature_initial_layer]

The absence of any start gcode defining the first or subsequent layer temps appears to cause the gcode reader to ignore subsequent temperature commands - representing the extruder temperature with whatever is defined in the start gcode, or 0 if otherwise not defined. The called in bed and extruder temps I am using are not registered by the gcode reader. When I send this print to the machine it heads up to the appropriate temperature defined by the filament profile and successfully prints. 

I tested adding M104 S150 before my start gcode and this caused the gcode preview to represent the entire print as 150C. This also had no adverse impact on printer execution of the print. Changing the gcode flavor had no impact on the gcode preview or machine behavior.

The temperatures reported in the gcode preview do not accurately reflect the temperature instructions present in the gcode.

EntwicklungsAnstaltGretsch, 2025-01-15:
I am experiencing a similar issue with the Anycubic Vyper profile. When slicing it actually does not set a hotend temperature at all, which had it fail to print for me until i figured that out.

Looking at the G-code at the start i find that it only heats up the heated bed but not the hotend:

```
; EXECUTABLE_BLOCK_START
M73 P0 R36
M201 X1000 Y1000 Z200 E5000
M203 X200 Y200 Z10 E60
M204 P1250 R1250 T1250
M205 X8.00 Y8.00 Z0.40 E5.00 ; sets the jerk limits, mm/sec
M190 S60 ; set bed temperature and wait for it to be reached
;TYPE:Custom
```

This is why i put in this line manually before the M190 command:
`M104 S220 ; set temperature`

Putting M104 before M190 makes both heat up at the same time

I also tried slicing the same model with prusa slicer and there it does the command properly.

When i import the gcode from PrusaSlicer into OrcaSlicer the temperature values are shown correctly in the G-code preview

So the issue is not the G-code viewer but the G-code generation

EntwicklungsAnstaltGretsch, 2025-01-15:
I started doing some digging and as it turns out it seems to be my modified printer profile. The stock profile adds the M104 command but after the M190 Command.

I found the Culprit for the Temperature in my case. I added the M109 command to the start G-Code, because i had the printer start printing with an almost cold hotend (90°C) which seems to remove the M104 command from the EXECUTABLE_BLOCK_START part of the G-code.

In my oppinion the M104 Command should be called before the M190 in order for the printer to heat up both hotend and bed at the same time. And for printers incapable of that have them do M190 and then M109 But don't have it be in order M190 and then M104 as that causes the print to start with a hotend that is too cold.

Another issue i have with the stock profile for the Anycubic Vyper is the Lack of the `G92 E0` command after the layer change. When i load the stock profile it shows me an error which sais this one is missing.

Here are the two generated G-codes named for their respective profile:

Had to rename files to .txt in order to be able to upload them to github
[TestObject_OrcaSlicer_STOCK.txt](https://github.com/user-attachments/files/18429836/TestObject_OrcaSlicer_STOCK.txt)
[TestObject_OrcaSlicer_MODIFIED.txt](https://github.com/user-attachments/files/18429843/TestObject_OrcaSlicer_MODIFIED.txt)

EntwicklungsAnstaltGretsch, 2025-01-15:
I ended up adding these two lines to my start G-code to get the desired behaviour:

M140 S[bed_temperature_initial_layer];Set Bed temperature
M109 S[nozzle_temperature_initial_layer];Set and Wait for Hotend temperature

EntwicklungsAnstaltGretsch, 2025-01-15:
#4337 is related

closed 2025-04-24 by github-actions[bot]