OrcaSlicer/OrcaSlicer #7583: Cut connectors misplaced

jeffdemeutter, 2024-11-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

13th gen intel core i7-1360P
16gb ddr4
nvidia RTX A500

### Printer

Bambu p1s

### How to reproduce

open Cut panel 
open `edit connectors`
select snap connectors with the following settings 
[image]
confirm the connectors
perform the cut
click yes on the `non-manifold edges` popup
[image]

### Actual results

[image]
[image]

### Expected results

[image]
[image]

### Project file & Debug log uploads

[connectorIssue.zip](https://github.com/user-attachments/files/17916387/connectorIssue.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

In one part of the object it seems to be placed fine and correct, but on the other object it gets offset.

I have also checked in BambuStudio, and it results in exactly the same.

I know that the object is currently too big for my build plate, but I first want to split the object as i am trying now, so I can cut it later into piececs that fit on the build plate.

Noisyfox, 2024-11-27:
I think the problem here is the model size changed after the repair, which lead to the offset of the connectors.
If you don't fix the model, you will notice this part:
[image]
(Note how the connectors are placed correctly)

However once you fix the model:
[image]

Noisyfox, 2024-11-27:
What you need to do is place the model perfectly flat on the plate, by using:
[image]

Then you can cut it perfectly without the need of model fix
[image]

jeffdemeutter, 2024-11-27:
I already was using lay on face, but I was laying it on the letters. I just tried to lay it on the background and cut it from there. This worked! 

I have made this model myself, and can ensure that the faces on the XY plane are all perfectly parallel. So when I lay it down on the letters, I would expect it to behave the exact same, no?

Noisyfox, 2024-11-27:
Weird... I lay on the latters too and it works somehow.

jeffdemeutter, 2024-11-27:
yes very odd, I just can't get it to work when its laid flat on the letters side. I tried multiple orientation around Z axis after. I tried flipping the connectors. Nothing seems to make it work for me.

closed 2025-03-05 by github-actions[bot]