OrcaSlicer/OrcaSlicer #7588: Add object to modifier conversion to right click menu

witorwitor, 2024-11-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Please add an option to convert objects to modifiers.

Common workflow involves designing a multipart object and exporting it as one stl. After loading this stl in the slicer and splitting it back into individual parts it would be great if some parts could be turned into modifiers like it is possible n prusaslicer.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

right click on the object/convert to modifier

DVSVIDEO, 2024-11-28:
The solution:
1. load the multipart object, stl file
2. press the "split into objects" button in the top menu
3. select and save the part you want to use as a modifier
4. select the object you want to modify, right click and add modifier---> load
5. select the saved part
It took me longer to write it down than it took you to do it.
I ask you and everyone else: please suggest extensions that cannot be solved in any other way  because the slicer is getting bigger and more complex

witorwitor, 2024-11-28:
I could also export object and modifiers as two separate STL files from CAD and then load the modifier STL. 
I know that it is possible to use loaded meshes as modifiers but both my and your methods take a lot of tedious steps. It would be a good life quality improvement to just select objects and then just one single right-click to convert them to modifiers like in prusaslicer. Of course, I understand that there are more important issues to solve...

VigilantGardener, 2024-12-03:
How would this differ from the existing "Change Type" functionality that's available in the context menu and in the objects section of process it that context menu?

DVSVIDEO, 2024-12-06:
Feel free to close this request, because VigilantGardener has shown that it's already done.

witorwitor, 2024-12-15:
I don't know what 'change type' functionality you are referring to. I just spent half an hour right clicking on different objects from my stl that was split to separate objects and I can not see the 'change type' anywhere...

witorwitor, 2024-12-15:
> The solution:
>
> 1. load the multipart object, stl file
> 2. press the "split into objects" button in the top menu
> 3. select and save the part you want to use as a modifier
> 4. select the object you want [...]

This does not work as after loading the saved stl's they are in a wrong position.

VigilantGardener, 2024-12-16:
> I don't know what 'change type' functionality you are referring to. I just spent half an hour right clicking on different objects from my stl that was split to separate objects and I can not see the 'change type' anywhere...

Change type only appears as an option when a single object out of an assembly is selected.

If you want to change a part into something other than a printable object, it needs to be part of an assembly.

VigilantGardener, 2024-12-16:
This is the change type window.  I got here by selecting the objects view and right clicking on the cylinder that's part of the assembly with the cube.

[image]

VigilantGardener, 2024-12-16:
You can also try exporting your models as step files.  When you bring those into Orca it will recognize all the separate objects individually AND group them in an assembly.  Then the change type option can be used to alter the properties of whichever objects you want.

witorwitor, 2024-12-16:
Looks like I achieved what I wanted after splitting my original stl to 'PARTS' not to 'OBJECTS'. After splitting to PARTS I was able to change part type to modifier and do what I wanted.