OrcaSlicer/OrcaSlicer #7599: Idea: Improving FDM print quality by keeping print times low with optimized outer layer heights

chropo8, 2024-11-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have an idea for making FDM prints look better without significantly increasing print time.
There is always a tradeoff between print quality and print time when it comes to layer heights. 
As I am printing and selling goods online, print time and print quality are both important to me. 

But so far, lowering the layer height for good optics always comes together with an increase in time.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

The concept is to divide only the outer wall in the z axis and print this outer wall in multiple steps with 1/2 or 1/3 of the regular layer thickness.
The inner wall(s) and structures can be printed with regular layer thickness to save time.
To make this work, the print order would likely need to be set to Outer/Inner, but that shouldn't be a problem— possibly except for critical overhangs.

For example: For a cube with a 20mm edge length, the print time increases by 84% when halving the layer thickness. With the solution described above, the print time would only increase by 23%—but the exterior would look just as good as the one that took 84% longer. A significant time advantage.

DVSVIDEO, 2024-11-27:
And how do you imagine that with sloping walls?

chropo8, 2024-11-28:
Good Point! There you would have to play around with the line thickness. I made a quick sketch by hand: 
[image]

closed 2025-03-06 by github-actions[bot]