OrcaSlicer/OrcaSlicer #7600: Slicer generates artifacts

voland696, 2024-11-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Generic Klipper printer

### How to reproduce

Slice model with manual tree support, 5 walls
[image]
[image]

### Actual results

artifacts on overhungs 
[image]
[image]

### Expected results

Artifacts are not expected.

### Project file & Debug log uploads

[artifacts.1.zip](https://github.com/user-attachments/files/17940059/artifacts.1.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

discip, 2024-11-27:
@voland696
Those are no artifacts.
It is what you instructed Orca to do by enabling **`Make overhangs printable`**.

Just disable **`Make overhangs printable`**. 😊

voland696, 2024-11-27:
OMG Thank you so much! I did not expect it will make it printable this way. My bad.

closed 2024-11-27 by voland696