OrcaSlicer/OrcaSlicer #7601: Keep model seams close to each other

Matszwe02, 2024-11-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When printing large models, especially with thin walls, sometimes seams are spread to opposite sides of the model, causing long travels just before printing outside perimeter, causing horrible artifacts

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

iterate over a handful of possible seam placements, count median (or standard average, I don't know which is more important) to all seam distances to choose the right seam combinations. 
In my example model this would end up with something like this:

[image]
where both seams are still on the same "corner sharpness level", but are much closer to each other, making hard to print (especially leaky) filaments look much nicer

### Describe alternatives you've considered

Manual seam painting on each model that has this issue

### Additional context

This could be added as another "seam position" option, to not conflict with the default behaviour, which could be more uniform due to not iterating over alternative placements

DVSVIDEO, 2024-11-27:
This can be easily done with an existing function. Seam painting in a few seconds.
No need to complicate an already increasingly difficult to understand slicer with another mathematical function, another menu item...
This is my private opinion.

Matszwe02, 2024-11-28:
> This can be easily done with an existing function. Seam painting in a few seconds. No need to complicate an already increasingly difficult to understand slicer with another mathematical function, another menu item... This is my private opinion.

Thanks for your opinion, but I still think slicer should be able to optimize each model without manual tuning. The same was with single-walled top infills, which is possible by doing that manually for each top layer, but slicer does that even faster than user can.

closed 2025-03-06 by github-actions[bot]