OrcaSlicer/OrcaSlicer #7608: Turn on Button

Andreas-strg, 2024-11-28:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Many users switch on the power supply of their printer with a wifi relay or similar

It would be easier / faster if you could operate them directly in Orca

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Button in the web interface tab that switches on the printer via http / mqtt or whatever

mrambossek, 2024-12-01:
would really like to see this as well. or some other kind of generic http call / mqtt sending facility with minimal UI.....

chwilfing, 2024-12-01:
Integration of Home Assistant, trigger an automation or run a service call would be totally great.

albznw, 2025-05-14:
This would be amazing