OrcaSlicer/OrcaSlicer #7611: Unexpected extreme (4000mm+) single overhang line generated with Arachne wall generator

aee0, 2024-11-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

Intel Core i7-7500u
16 GB RAM

### Printer

BIQU B1 and/or Voron Legacy

### How to reproduce

Load this STL file for the Dragon Burner: Alternative_Voron_Mounts/Modified_Mounts/V1_V2/STLs/MGN12H/MGN12H_X_Carriage_Lite_Lugs.stl (see https://github.com/chirpy2605/voron/blob/1c552edef49fac5af05f5b09b533ab4d0699a1dd/general/Alternative_Voron_Mounts/Modified_Mounts/V1_V2/STLs/MGN12H/MGN12H_X_Carriage_Lite_Lugs.stl)

Use the Arachne wall generator and slice the file using either a BIQU B1 profile or a profile based on the Voron Trident profile with 0.2 mm layer height.

### Actual results

On layer 88, there is a single overhang wall generated going out to over 4000 mm off the print area.

See screenshot below, with GCODE line highlighted. Notice the blue line (labeled as line type "overhang wall") going off the right edge of the build plate. 

[image]

### Expected results

There should be no overhang wall line here. 

### Project file & Debug log uploads

[orcaslicer-4297mm-overhang-bug-MGN12H_X_Carriage_Lite_Lugs.zip](https://github.com/user-attachments/files/17953119/orcaslicer-4297mm-overhang-bug-MGN12H_X_Carriage_Lite_Lugs.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

If slicing with the "Classic" wall generator, the issue does not exist:

[image]

discip, 2024-11-29:
@aee0
If I'm not mistaken this will be fixed by this #7330.
Please try this commit https://github.com/SoftFever/OrcaSlicer/actions/runs/11650728839.

discip, 2024-11-29:
@aee0
If I'm not mistaken this will be fixed by #7330.
Please try this commit https://github.com/SoftFever/OrcaSlicer/actions/runs/11650728839.

aee0, 2024-11-30:
@discip yes you're right, that PR seems to fix it! Thanks!

Should I close this issue or wait until the PR is merged?

discip, 2024-11-30:
@aee0
It's up to you.
If it is left unattended for 97 days, it will be closed anyway. 😊

aee0, 2024-12-01:
Thank you! I'll close it since the fix is on its way 😀

closed 2024-12-01 by aee0