OrcaSlicer/OrcaSlicer #7614: Reduce Print Temperature for short layer times.

Finnitio, 2024-11-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Achieving good printing quality with short layer times.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would like a layer time dependent feature, where depending on the layertime the slicer will automatically reduce the Extruder Temperature.

### Describe alternatives you've considered

Alternative feature is to reduce print speed in order to increase layer time but i have realised that below a certain size this change is not sufficient.

### Additional context

[image]
Relatively short layer times for this part.
[image]
Print temperatures form left to right: 235 °C, 215 °C and 205 °C.
I think the differences in surface quality are evident. This is printed with pet g, but I observed similiar problems with pla.

vgdh, 2024-11-29:
It would be nice, but not clear how it affects the surface quality.

kylek29, 2024-11-29:
This has been suggested a few times as a way to deal with other surface imperfections like glossy flashing (where longer layer times result in more matte finishes and short result in glossy). I think the discussion was that temperature inertia (the temp probe is lags what the nozzle sees) would be too much of a problem given the short time it would need to ramp down.

I still wouldn't mind having it as an experimental feature for us to toy with and see if it is viable at all.

ShaneDelmore, 2024-11-29:
I would love to see something like this in orca someday but if you want to play with these ideas now this postprocessor might help.  https://github.com/sb53systems/G-Code-Flow-Temperature-Controller It is not mine, just a neat project I saw.

Finnitio, 2024-12-06:
Wow that post processing script is impressive, I will look into it but share a siliar theory.