OrcaSlicer/OrcaSlicer #7617: Unexpected movements during tool change when using extruder offsets in idex mode and using purge tower

JuanR3D, 2024-11-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

R3DIMENSION IDENTITY 

### How to reproduce

Configure a multi-extruder printer with defined offsets between extruders in OrcaSlicer.
Generate a model with tool changes and enable the purge tower.
Analyze the generated G-code or run the print.
Observe how the nozzle moves to unexpected points before purging on the tower (in my case, almost origin of the bed)

### Actual results

Currently, during a tool change:

The new tool (T0 or T1) is activated.
The slicer generates movements to specific fixed points on the print bed, seemingly related to the defined offsets between the extruders.
Only after these movements does the nozzle move to the purge tower to perform the purge and continue printing.
These additional movements seem unnecessary and unrelated to the printing process or the purge tower

### Expected results

When performing a tool change, the following flow should occur:

Deactivate the current extruder (e.g., T1).
Activate the new extruder (e.g., T0).
Move directly to the purge tower.
Purge on the tower and continue printing without unnecessary movements outside the tower

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[image]
[image]
[image]
[image]

JuanR3D, 2024-12-13:
hi @ink93  my purge tower has these settings (i guess you want the circle in order to make it more rigid):
[image]

About the zoffset in T1... does it apply only when it purges or also when it prints the piece?

ink93, 2024-12-13:
I have not set anything at T1 Z offset. Everything as it was. Since the Z offset runs via an Idex pin in Klipper. It does the Z offset every time at Start_Print.

I can send you my settings from OrcaSlicer.

closed 2025-03-21 by github-actions[bot]