OrcaSlicer/OrcaSlicer #7620: Not possible to add a pause during print

Takao-7, 2024-11-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

15.1.1 (24B91)

### Additional system information

Mac Book Pro 2021

### Printer

Bambulab A1

### How to reproduce

1. Add a model to a project. This can either be one material or two
2.Slice a plate 
3. Move the layer slider to any position
4. Try to right click on it

### Actual results

- If you used a two-finger tab, the slider will jump down by two layers
- If you did a right click on the trackpad, nothing happens

### Expected results

A menu opens where you can add a pause

### Project file & Debug log uploads

[debug_Fri_Nov_29_23_04_43_49472.log.0.zip](https://github.com/user-attachments/files/17963270/debug_Fri_Nov_29_23_04_43_49472.log.0.zip)
[Filament Lagerung - Labels.3mf.zip](https://github.com/user-attachments/files/17963271/Filament.Lagerung.-.Labels.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

I don't have an AMS

Takao-7, 2024-12-01:
This issue doesn't happen on Windows. Looks like it's Mac OS specific, or even trackpad-only.

Note that the "trackpad camera" option doesn't effect the issue.

badVibes--, 2024-12-07:
Such a frustrating bug! I wish I could disable "scroll wheel" on that slider. The problem is trackpad only I presume. As soon as two fingers touch the trackpad it starts scrolling. This is problematic because we have "two finger secondary click" on trackpads. 

This might be a regression, I don't remember having this before.

They way I worked around it was to go and temporarily change the trackpad secondary click from "two finger" to "bottom right corner". This way you can click on the slider without touching the trackpad with two fingers.

I hope this gets fixed :/.

Takao-7, 2024-12-09:
The issue also happens when only one finger touches the trackpad and I press it in the right side for a right-click

grooverkm, 2024-12-27:
I have the same problem, MacBook running Sequoia. The addition of a keyboard shortcut would be helpful.

closed 2025-04-04 by github-actions[bot]