OrcaSlicer/OrcaSlicer #7625: layer width not correct

Avatarsia, 2024-11-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 & 2.1.1

### Operating System (OS)

Windows

### OS Version

Win11

### Printer

on any printer

### How to reproduce

create an object, check preview line width.
open the exported gcode and recheck line width

### Actual results

the linewidth is actually not what it should be. 
arachne dont effekt it, scarf joint have a litte effect

### Expected results

all settings of layerwidth set to .4 or .5, the even top layer is not consitent in .4 or .5 of a voron cube

### Project file & Debug log uploads

no log

### Checklist of files to include

- [X] Log file
- [X] Project file

Avatarsia, 2024-11-30:
clasic no scarf joint
[image]
classic with scarf joint (2picutres)
[image]
[image]

preview
[image]

all have been set to 0,5mm layerwidth

Noisyfox, 2024-12-01:
Could you share a 3mf file that can reproduce this issue?

Avatarsia, 2024-12-01:
you could simple try it your own version, 
it's not based on my mashine, i was hinted to this by another user. 
[voroncube.zip](https://github.com/user-attachments/files/17969777/voroncube.zip)
include 3mf and resulted stl

Avatarsia, 2024-12-06:
> Could you share a 3mf file that can reproduce this issue?

could you reproduce it ?

blalor, 2025-01-13:
I'm having this same problem with OrcaSlicer v2.2.0; it's extremely visible on the first layer during printing, but looking more closely at the gcode with PrusaSlicer it appears to be an issue in many places.

[Cube_0.3mm_PLA_RatRig V-Core 3 400_28m56s.gcode.zip](https://github.com/user-attachments/files/18392438/Cube_0.3mm_PLA_RatRig.V-Core.3.400_28m56s.gcode.zip)
[Cube.3mf.zip](https://github.com/user-attachments/files/18392439/Cube.3mf.zip)

PrusaSlicer's gcode viewer (v2.9.0) shows the lines spaced far apart (which is how it looks in real life):
[image]

And also shows that the width is closer to 0.5mm, instead of the ~1.0mm that OrcaSlicer was configured to use, and which OrcaSlicer seems to think it's emitting based on the comment in the gcode:
[image]

OrcaSlicer's preview shows the lines being close together:
[image]

closed 2025-04-21 by github-actions[bot]