OrcaSlicer/OrcaSlicer #7632: Conditional G-Code causing Unexpected Syntax Error Incorrectly

Construct3D-AM, 2024-12-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Various systems

### Printer

Construct 1 XL (RepRapFirmware)

### How to reproduce

What works:
**M109 S[first_layer_temperature];** set extruder temp

The above works as intended, it sets the first layer temp to the sliced G-Code with no issues.
The goal was to use conditional G-Code to automatically set a waiting temperature 60C bellow the printing temp.

For RepRapFirmware the code will look like this:
**M109 S{[first_layer_temperature] - 60};** set extruder temp

Objects that are inside curly brackets are treated as math operations, The intended outcome would be to inject the "first_layer_temperature" value into the curly brackets.

When testing via a macro file on the printer itself, the code works. but when slicing via Orca it throws an error of unexpected syntax because of the use of curly brackets

### Actual results

[image]

Failed to generate G-code for invalid custom G-code.
machine_start_gcode Parsing error at line 6. Unknown syntax error
M109 S{[first_layer_temperature] - 60}; set extruder temp
            ^
Please check the custom G-code or use the default custom G-code.

### Expected results

M109 S160   ;This is if the initial material temp would be 220C - 60C = 160C

### Project file & Debug log uploads

Literally nothing to provide, Logs don't show errors in slicing, and problem can be replicated easily

### Checklist of files to include

- [ ] Log file
- [ ] Project file

henrivdr, 2024-12-04:
I think it should be 
M109 S{first_layer_temperature - 60};

Construct3D-AM, 2024-12-04:
Unfortunately by dropping the square braces, then the slicer is unable to inset the [first_layer_temperature].
This is because when the slicer is compiling the code, the square braces act as a start and end point for a single variable, which is expecting a single string to be used.

Using [first_layer_temperature - 60] fails as there are now 3 strings, and the "-" and "60" are unexpected
Replacing the square braces like so {first_layer_temperature - 60} will correctly signify a conditional Gcode operation. however the slicer now doesn't know what "first_layer_temperature" is because it uses the square braces as variable identifiers.

Examples tested:
machine_start_gcode Parsing error at line 6: Referencing a vector variable when scalar is expected
[image]
machine_start_gcode Parsing error at line 6. Expecting tag literal-char
[image]
machine_start_gcode Parsing error at line 6. Unknown syntax error
[image]
machine_start_gcode Parsing error at line 6. Expecting a legacy variable expansion format
[image]

What does work is "M109 S[first_layer_temperature]{ - 60};" however this then fails on the machine as M109 S220-60 is not correctly formatted as it requires {} to cover the whole operation

I believe the bug is solely due to the slicer not expecting a { directly after the S

henrivdr, 2024-12-04:
The index into the vector variable is missing. Try 
M109 S{first_layer_temperature[0] - 60};

Construct3D-AM, 2024-12-05:
I think that may have worked!.  Initial slicing has passed. and the exported G-code shows the correct value in the M109 Snnn

henrivdr, 2024-12-26:
@SoftFever I think this one can be closed. 
Correct @Construct3D-AM ?

Construct3D-AM, 2024-12-28:
Ahh yes, I thought I already closed this as completed. My bad

closed 2024-12-28 by Construct3D-AM