OrcaSlicer/OrcaSlicer #7639: Using the "fan sped-up time" feature leaves the fans on during purge which leads to poop piling up in the chute on multicolor prints

shyblower, 2024-12-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux, macOS, Windows

### OS Version

all

### Printer

Bambu Lab X1C

### How to reproduce

Set "Fan speed-up time" in the printer profile to any value other than 0

### Actual results

M106 P1 S0
M106 P2 S0
from the filament change g-code of the printer profile will be removed from the produced g-code, effectively leaving the fans on during purge, which will, depending on the previously set fan speed, fail to form a compact poop but produce a rather large and stringy ball which cannot be purged and thus pile up in the chute effectively filling it up after a few filament changes resulting in either pausing the print when chute pile up detection is enabled or lead to lots of issues.

### Expected results

M106 P1 S0
M106 P2 S0
inside the filament change g-code should not be touched by the "Fan speed-up time" feature, so that the cooling fans are stopped during filament purge to ensure that the poop will be as compact as possible and can be disposed of by the chute emptying mechanism.

### Project file & Debug log uploads

[fan_speed-up_time_issue.zip](https://github.com/user-attachments/files/17979474/fan_speed-up_time_issue.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

#2144

pennystockss, 2024-12-03:
Just happened today to a project, forgot to select my custom profile and left one color on a default Bambu PLA profile so I hopped on here to check my old issue #2144  from, 14 months ago lol, which looks like it's been closed for being stale.

It really bums me out because I'd like to use the advanced fan features but can't because it messes up all my prints. I just set the fan to 100% on all my profiles 🤷‍♂️.

philltran, 2025-03-04:
> Orca bot: this issue is stale because it has been open for 90 days with no activity.

This issue is still present for me

closed 2025-06-11 by github-actions[bot]