OrcaSlicer/OrcaSlicer #7645: OrcaSlicer 2.2.0 crashes without notification when rendering specific model

aheil, 2024-12-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

OS Microsoft Windows 11 Pro
Version: 10.0.26100 Build 26100
CPU: Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz, 2201 MHz, 6 Kern(e), 12 logische(r) Prozessor(en)
BIOS-Version: American Megatrends Inc. N.1.09, 11/29/2018
Emomry (RAM): 16.0 GB

### Printer

Anycubic Kobra Neo 2

### How to reproduce

Open the m3f file and select preview or slice plate.

### Actual results

While slicing, the program crashes end exits without notification.

### Expected results

The model gets sliced and is ready for printing.

### Project file & Debug log uploads

[2024-03-12-CrashExample.zip](https://github.com/user-attachments/files/17989580/2024-03-12-CrashExample.zip)
[crash_Tue_Dec_03_09_10_30_2.log](https://github.com/user-attachments/files/17989581/crash_Tue_Dec_03_09_10_30_2.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

There is a similar issue, #4461 with the same effect but with other reasons.

aheil, 2024-12-03:
I was able to find an workaround. 

- Right click the model
- Select "Simplify Model"
- Choose the settings as in the screenshot 

[image]

Afterwards, OrcaSlicer does render the model without crashing.

This behavior is absoutely reproducable. I was able to crash OrcaSlicer with an .stl and an .step file of the same model.

Ergonomicmike, 2024-12-03:
I tried it on Orca v2.2.0. It didn't crash. (Win 10) But after it completed, I received an error that the G-code went beyond the boundary.

There has been a commit that is said to have fixed this G-code issue. I have to go now but will try a Nightly later.

MartyFufkin, 2024-12-04:
2.2 is crashing at 80 percent slicer for me all the time when using the Arachne engine but works fine with Classic.  Crashes computer completely and sometimes to BSOD. Build volume well within plate.  Happy to provide more info is specifics needed.  I can provide the stl as well.

Ergonomicmike, 2024-12-05:
To confirm from my end, your model slices fine for me using my Printer, Filament and Preset with the Arachne engine.

However, it does crash Orca when I use YOUR Printer, Filament and Preset per you m3f. I saw a message flash on the screen about a memory violation.

Unfortunately Orca doesn't allow me to compare my presets to yours because my Presets are all tied to my Printer. But apparently the problem is in one of those.

Ergonomicmike, 2024-12-05:
@Azio-Pantheon 
[2024-04-12-CrashExample working with my Ender 3 v2 Profile.zip](https://github.com/user-attachments/files/18016478/2024-04-12-CrashExample.working.with.my.Profile.zip)

Ergonomicmike, 2024-12-05:
@Azio-Pantheon @aheil I have observed that sometimes issues reported here as bugs are caused  by broken models. 
Maybe the OP will run his model through the Formware B.V. online stl repair tool?

aheil, 2024-12-05:
@Ergonomicmike I absolutely see your point. The bug is not about the model, the bug is that OrcaSlicer does crash while slicing the model. Said that, my expected  behavior of OrcaSlicer would be at least a (meaningful) error message and not crashing at all. If the model cannot be sliced, this would be fine for me as long as the slicer does not crash without any further notification.

Ergonomicmike, 2024-12-05:
FWIW, I took your .3mf and told Orca to make an .stl out of the model. I ran the model thru the online .stl Repair Tool. It reported no errors. (Sreenshot.)

[image]

Having said that, I'm told by the experts here that there is another tool for checking .stl's (mesh something) that I don't have.

Here's the report that Windows throws when your model crashes when I use your Printer, Filament, Process.

[image]

closed 2025-03-13 by github-actions[bot]