OrcaSlicer/OrcaSlicer #7649: Crashes when language set to en on a system without en_GB support

n3oney, 2024-12-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

NixOS unstable

### Printer

Ender 3 Pro

### How to reproduce

Set your language to english (`en` in OrcaSlicer.conf)
Launch the slicer

### Actual results

Observe this error:
Switching Orca Slicer to language en_GB failed.
You may need to reconfigure the missing locales
[image]

### Expected results

It launches.

### Project file & Debug log uploads

no related logs

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Seems to be the same issue as this:
https://github.com/prusa3d/PrusaSlicer/issues/3368

TerraRoot, 2025-01-23:
Getting the same error, not sure where en_GB comes from since my system is set to en_IE. edited the OrcaSlicer.conf to en_IE from en_US and it works.

Perhaps the bug is somewhere in the locale detection?
what's your locale @n3oney ?

n3oney, 2025-01-23:
My locale is en_US.UTF-8.

HerrCules, 2025-01-27:
Hey there, i also get this error, but therefore I get: 
use system SSL certificate: /etc/ssl/certs/ca-certificates.crt
To manually specify the system certificate store, set the SSL_CERT_FILE environment variable to the correct
CA and restart the application
Do you want to continue?

Now the problem is, that I can't change the language in the .config file, because it doesn't exist. I run the first time Orca, probably that is the reason.
I use Debian and my OS-Language is German.
How can I fix that without the .config-file?

HexSegfaultCat, 2025-02-15:
I had the same issue on Arch with my system's locale set to `pl_PL.UTF-8`. In my case I just needed to edit the file `/etc/locale.gen` and uncomment the line `en_US.UTF-8 UTF-8` and then ran `sudo locale-gen`. After that the `en_US` locale shows up in `locale -a` and the slicer works.

Smoerrebroed, 2025-03-23:
I came across this one today as well running version 2.3.0. The inconvenient thing is that you have to generate en_GB in the ISO variant. en_GB.UTF-8 will not suffice. And between us: I'd think it's far more likely that someone uses en_US (in any variant) rather than en_GB, so why not make that the default for English?

semiAquaticOtter, 2025-05-05:
i fixed it by un-commenting "en_GB" in /etc/locale.gen. and then ran "sudo locale-gen". 

hope this helps anyone with this problem 

(arch linux + hyprland)

flaviut, 2025-08-03:
Looks like orca-slicer [shells out to `locale -a` to handle this case](https://github.com/prusa3d/PrusaSlicer/blob/master/src/slic3r/GUI/GUI_App.cpp#L2365C23-L2365C29), and if that command fails, it silently does nothing. I don't know Nix, but is it possible to add the `locale` command to orca-slicer's `PATH` in the nix packaging?

Also mentioned here: https://github.com/NixOS/nixpkgs/pull/430171#issuecomment-3148529629

Dustie, 2026-03-28:
I have the exact same problem in 2.3.2 (and nightly) in Windows 11.

gizmo-boss, 2026-04-02:
> I have the exact same problem in 2.3.2 (and nightly) in Windows 11.

Same problem