OrcaSlicer/OrcaSlicer #7654: 2.3.0 Dev nightly - Support interface in mid air with no support under it.

tlhintoq, 2024-12-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.3.0 dev

### Operating System (OS)

Windows

### OS Version

Win10

### Additional system information

11th gen i9 with nvidia

### Printer

Bambu

### How to reproduce

See attached project.  Look at the settings.

### Actual results

Support interface but no support under it.
[image]

### Expected results

Don't make support interfaces in mid air.

### Project file & Debug log uploads

[UpperArm.bambu.orca.zip](https://github.com/user-attachments/files/18002668/UpperArm.bambu.orca.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]

Hammerfest, 2024-12-05:
Noticed this as well as of late, in addition to supports starting at the bottom, but not existing at the top (the reverse of this report)

igiannakas, 2024-12-06:
Is this reproducible in 2.2 release?

tlhintoq, 2024-12-08:
> Is this reproducible in 2.2 release?

Its been doing for several releases... Then the last ticket for it went dead for lack of activity.  maybe a new ticket will get some love.

tlhintoq, 2024-12-11:
I really appreciate the time and effort you put in to narrowing down the problem and isolating it to just a specific setting that works.

> Tree organic looks okay

That's a shame.  `Tree | Organic` has enough other issues that its not reasonable to use in a lot of cases.
Single threaded calculation - So REALLY slow to slice. And with a Giga printer that's got big items it can be forever or just outright crash.
Frequently produces supports with big shifts and supported sections causing them to eventually collapse/fail.
Internal to the trunk there's almost always unsupported artifacts that become a mess, grab the nozzle, cause the support to catch and be pulled apart or catch and cause a layer shift.  Basically they're just not a viable option.
[image]
The tickets for these problems never get addressed and eventually auto-close, but its still happening in 2.3.0-dev
https://github.com/SoftFever/OrcaSlicer/issues/6259

tlhintoq, 2024-12-19:
So I tried doing the `Tree|Organic` option.
There's more flowing rocks than on the planet Pandora.
[image]

tlhintoq, 2024-12-28:
Anyone got a hint, or clue or something... a magic combination of settings that stops this?  because printing in mid air results in big blogs and eventual print failures.
Is there some way to convince orca to put some thing under it... convince it it needs infil there or something?
[image]

tlhintoq, 2025-01-22:
{nudge}  Any progress on this?  It totally eliminates the use of some of the best support options which is a real shame.

https://github.com/user-attachments/assets/d3ffa421-b3c5-4a28-a27d-dea51982a15a

SleepyBread, 2025-01-24:
Same issue over here. If we keep this ticket open they might hopefully give it some attention one day

[image]

michaeljgolden, 2025-01-27:
I just ran into this problem.  Noticed while printing  :(

It tried to print this floating support:
[image]

Here is another.  This one isn't as bad, but it still won't print these odd shapes in mid air.
[image]

Note:  This print has like 5+ of these in it.

Tarakesh83, 2025-01-28:
Tree Slim can be "fixed "by setting Top Interface Layers to 0 as recommended  in the Popup when you switch to it.

2 Top Interface Layers, 0.14 Distance
[image]

0 Top Interface Layers,  0 Distance

[image]

Is it good? No, but its better than before

igiannakas, 2025-04-29:
@Noisyfox any ideas? 🤔

tlhintoq, 2025-09-06:
Not yet fixed

tlhintoq, 2026-10-30:
{bot nudge}