OrcaSlicer/OrcaSlicer #7657: Make Debian/Ubuntu native packages

Centuriondan, 2024-12-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Flatpaks... hate them... bloat.. slow to build... security risks from old versions of libs etc...

Debian based distro's (like Devuan) should be able to utilise standard library version and build just Orcaslicer in the normal debian style tool chain.

I've checked the library versions and have started doing minimal patching to get Orcaslicer to build without too much hassle - and even finding a few minor bugs along the way (once I can verify the issue isn't simply a matter of library changes, I'll post bug reports and pull requests with the fixes.

One notable requirement is the need to upgrade to GTK3 (in my case GTK3.2)

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I ultimately should be able to apt-get install orca-slicer and have it install in a few seconds...

I've checked the library versions and have started doing minimal patching to get Orcaslicer to build without too much hassle - and even finding a few minor bugs along the way (once I can verify the issue isn't simply a matter of library changes, I'll post bug reports and pull requests with the fixes.

Requires an upgrade to GTK3 (in my case GTK3.2)
Move to distro supplied OpenCASCADE (I have patches that make this build but need tweaking to probably not break flatpak builds)

### Describe alternatives you've considered

1) Just put up with flatpaks...  insecure, fat and libs aren't updated very quickly...
2) Build a flatpak from source using as many local libraries to minimise the number of dependencies broken.
3) Get Windows (tm)  ..... NEVER!!!
4) Do it myself.... well... yeah I'm already well on the way to doing this :-)

### Additional context

Current status... Compiles against all the distro libs detected, including OpenCASCADE.
Fails on INSTALL part of the build 
TODO: I need to understand all the built components to determine whether one package is sufficient or it could/should be split into multiple deb packages (for optional parts).

KaiMartin, 2026-03-01:
Works for me, too. Thank you for putting up the effort!
My system is on Debian/testing aka, forky.

dsptech11-prog, 2026-09-03:
This is awesome. Thanks for making this possible. Was getting so tired of using Flatpaks. Is there a way to add your github releases as an Apt source?