OrcaSlicer/OrcaSlicer #7663: AMS showing ASA filament in all slots when machine is set to Generic PLA for all slots

iron-bro, 2024-12-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Intel NUC Hades Canyon
Processor	Intel(R) Core(TM) i7-8705G CPU @ 3.10GHz   3.10 GHz
Installed RAM	16.0 GB
Device ID	B949929F-0F5F-4EEF-A0E4-E3AB396128D8
Product ID	00326-10020-35488-AA362
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

### Printer

Bambu X1C 0.4mm 

### How to reproduce

1. Select Generic PLA as filament material in Prepare Tab
2. Adjust slice settings as needed
3. Click 'Slice Plate'
4. Click 'Print Plate'

Issue happens at preview screen before hitting send.

5. Select Device Tab
6. AMS slots read "ASA" in each slot
7. Click on a slot to edit filament type
8. Slot reads "Generic PLA" which is what had been selected on the printer prior to slicing
9. click reset
11. Click ok when prompted "are you sure you want to clear the filament information?"
12. filament type resets
13. click edit button for that same slot
14. select Generic PLA and colour
10. Click ok when prompted "are you sure you want to clear the filament information?"
11. filament type resets
12. click edit button for that same slot
13. select Generic PLA and colour and click confirm

Filament gets automatically changed to ASA

14. Bang head against wall in utter frustration lol

### Actual results

[image]
[image]

### Expected results

Filament should show PLA in AMS slot and allow print to go forward.

### Project file & Debug log uploads

[Orca Slicer Issue - Project File (Geometric Container) .zip](https://github.com/user-attachments/files/18014937/Orc
[debug_Tue_Dec_03_18_46_55_24948.log.zip](https://github.com/user-attachments/files/18014946/debug_Tue_Dec_03_18_46_55_24948.log.zip)
a.Slicer.Issue.-.Project.File.Geometric.Container.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

This has happened before and it seems to happen randomly at times on this device and other windows 11 devices I have used Orca Slicer on. 

I have multiple X1C so I tried to send this print to the others and the same issue happens

iron-bro, 2024-12-05:
Update:

I tested a few things to try and pinpoint the issue. It seems like this issue only happens when Generic PLA is selected. I switched the filament type over to Bambu PLA Matte and it worked but everytime I go back to Generic PLA it switches it to ASA. 

This issue happens no matter what Bambu X1C I switch to (I have 6). Changing to other filament types like PETG works just fine its only when trying to use Generic PLA. 

The weird thing is that Orca slicer was working just fine for me over the last couple of weeks then today for some reason it does this. 

I tried multiple project files and the same issue happened across all of them. Tried a fresh install of Orca, still same issue. Tried restarting PC but still got the same issue.

void1972, 2024-12-24:
Same issue on A1 with AMS Lite, since I installed the new Firmware v01.04.00.00 on my Printer.
Every PLA switches to ABS.

jordanprazak, 2024-12-29:
Same issue. Generic PLA shows up as ASA in the device menu. This makes it so that I am unable to select PLA as a filament option when sending a print to the printer. 
Printer: BambuLab X1C with AMS
OrcaSlicer Version: 2.2.0
OS: Windows 10 Pro 22H2

KeluDev-Code, 2025-01-09:
Same issue. Generic PLA shows up as ABS in the device menu.

Printer: BambuLab A1 with AMS
OrcaSlicer Version: 2.2.0
OS: Zorin OS 17.2 Core ( powered by Linux kernel 6.8 from Ubuntu 24.04 LTS)
[image]

vfbank, 2025-02-11:
This issue is a bug that occurs when adding other printer specific model profiles (e.g. Flash Forge 5M PRO or Creality Printer K1 MAX). Currently, the only way to fix this is to delete all the Orca files in your APPDATA Orca folder.

**-WARNING!- Do not attempt to delete the USER APPDATA file unless you are an expert. All your printer settings will be lost!**

iron-bro, 2025-02-12:
@vfbank thanks for clarifying this!

ok-67, 2025-02-12:
I also see the same error, P1S with AMS. Generic PLA shows up as ASA. Running Linux version 2.2.0

I find a lot of Orca configfiles in my system, located in either "./.local/share/orca-slicer" or "./.config/OrcaSlicer" with several subdirectories. So where to delete?

closed 2025-05-21 by github-actions[bot]