OrcaSlicer/OrcaSlicer #7671: Speed/Extrusion Smoothing With Arc Fitting

Tasmanian666, 2024-12-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Hello,
According to issue #4216, Arc Fitting doesn't work if Extrusion Rate Smoothing is enabled.
However, Bambu Studio has a parameter called "Smooth Speed Discontinuity Area" which allows to achieve similar speed smoothing without giving up Arc Fitting.
Is there a plan to implement this feature or any other speed smoothing setting compatible with Arc Fitting?

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Implement "Smooth Speed Discontinuity Area" from Bambu Studio or another speed smoothing setting compatible with Arc Fitting.

bingzhuyeyou, 2024-12-08:
I noticed this problem too, and I couldn't stand the noise caused by printing arcs without turning on arc fitting, so I had to turn off extrusion rate smoothing.

bingzhuyeyou, 2025-03-09:
作业以收到，这属于自动回复，，，，秉烛夜游

CCS86, 2025-05-17:
I wish this issue would get some more attention.

Arc filtering is pretty important on the Bambu printers.