OrcaSlicer/OrcaSlicer #7679: After running 2.2.0 portable all my custom printers/configs/profiles are gone from previous 2.1.1 installer version

marcinviz, 2024-12-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1 -> 2.2.0

### Operating System (OS)

Windows

### OS Version

10

### Printer

Bambu Lab A1/ Custom Marlin printer with octoprint on PI

### How to reproduce

I downloaded and run 2.2.0 portable version to check for new stuff and test new Bambu Lab A1 printer. I had previous 2.1.1 version installed via installer, it was configured with custom made marlin printer and some self-made profiles, gcodes etc. After launching 2.2.0 portable, I have added Bambu Lab A1 printer and logged in on home tab to connect to it. Now i discovered that all my previous made printers, profiles, configs are gone. Only ones that was made after adding Bambu Lab printer are available. Cant find the previous ones in any folder. I tought that running a portable version uses local folder filesystem, not the general system one that overrides the old one...

### Actual results

Custom made profiles and configs are gone and nowhere to be found.

### Expected results

Portable version should not use original installed version folder structure to override previous made user configs/profiles. No warning was visible to avoid this.

### Project file & Debug log uploads

Don't have any

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Ergonomicmike, 2024-12-07:
Whether this is a bug or intended behavior, I don't know. 

But you might be able to find your previous presets in Users/[Your account folder]/AppData/Roaming/OrcaSlicer/

IIRC, there used to be version folders there. I don't see any in mine now. But I suggest making a copy of the OrcaSlicer folder as a backup and then see if you can find your old presets/profiles in either a .json file or OrcaSlicer/conf - Copy.bak

e-caliskan, 2024-12-13:
I have the same problem. When backing up, the printer that the user created is not in the backup options. Sad! I cannot see my own printer profile in the backup options and it does not back up.

Manuel backup > %userprofile%\AppData\Roaming\OrcaSlicer\user\

[image]

closed 2025-03-21 by github-actions[bot]