OrcaSlicer/OrcaSlicer #7680: Brim Auto puts a brim on EVERYTHING, even large parts with very large contact areas on the build plate that clearly don't need it.

mytechguyri, 2024-12-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11 Pro Insider Preview (10.0.27754 Build 27754)

### Additional system information

AMD Ryzen

### Printer

Bambu Lab P1S

### How to reproduce

Every print, no matter how large or how much contact with the build plate, if Brim Type is set to Auto, it will put a brim.

### Actual results

Brim on every print unless "no brim" is specified

### Expected results

Brim on prints with a small build plate contact area, or very small parts where a brim is likely necessary to ensure part adhesion, and no brim on large parts with a large build plate contact area that likely do not need a brim.

### Project file & Debug log uploads

na

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ShaneDelmore, 2024-12-08:
I just tested and found the same issue on M1 Mac in 2.2.0.  I added a cube primitive to the build plate with brim set to auto, then made the cube 150x150x10mm and got a brim where I would not expect one.

[image]

pgifford, 2024-12-11:
> Brim on every print unless "no brim" is specified

Every print? I don't get a brim on 25x25x25 cube but I do on a 150x150x150 cube.

I don't think that "does not meet your expectations" is a bug, especially since the brim is easily disabled. This seems like an enhancement request.  Either way you should provide a formula or algorithm describing what you think is a better method for determining when a brim is required.  A single data point ("150x150x10mm") is not sufficient.

Dan203, 2025-01-23:
I am also running into this issue and it's ruining my prints. I know it can be disabled but I just switched over from Bambu Studio and I'm not use to having to do that so I've had more than one print end up with unneeded brims that were so tightly stuck into the grooves of a print that they were impossible to remove cleanly so I ended up having to reprint. 

I'm curious why there is a difference with this setting compared to Bambu Studio? Was there an intentional change to the algorithm or is Orca just out of sync with the Bambu Studio branch in that area?

closed 2025-05-01 by github-actions[bot]

juszczakn, 2025-09-05:
I'd like to bump this. It seems like Auto is putting brim on pretty much every print that I make, even when it's laughably unnecessary. I'd like to keep it on auto, to make sure I don't accidentally forget to turn it on, but the current parameters it's using seem baffling.