OrcaSlicer/OrcaSlicer #7681: Support Interface Line Width

movinimage, 2024-12-07:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

There is an old, closed issue related to this (Line width for Support Interface and for Overhangs #102). That issue never fully fleshed out why the improvement would be helpful. It seems like a re-do might be helpful. If not, we can merge the issues.

Adjusting support interface line spacing is a good way to make support easier to remove. But, it has some downsides. If you are printing with a large nozzle like .6, and thin layers, you end up with a very flat filament cross section and a huge contact patch. Using line spacing to mitigate this often results in supports that are to far apart and cause drooping in between them. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

If "Support Interface line spacing" was independently adjustable, you could have closely spaced support lines that had small contact patches and were easier to remove.

I made the changes larger for visual effect in the following pictures.

Support=.7, Spacing=.4
[image]

Support=.3, Spacing=.2
[image]

### Describe alternatives you've considered

You can currently adjust the "Support" line width, and this will also change the interface line width. But, doing this especially with tree supports or combined layers can make the support structures very iffy (setting a "Support" width of .4 on a .6 nozzle with combined layers is a recipe for underextrusion and poor support structures). Keeping them separate allows robust support structures with more delicate interfaces.

lordmorgul, 2025-12-04:
I want it!

ArztVielfrass, 2026-11-16:
Cura has this setting. It's the one setting that sets apart the support quality of Cura and Orca.  
You need the supports to be thick so they can support the melted layers on top of them. However you also need the interface layer to be as thin as possible so that they can peel off the layers they are supporting very easily. If they are thin enough, certain sticky filaments can knock the support from the build plate, making the lines thick prevents this. The interface distance is a tradeoff, and a single layer height provides the best overhang quality but can be difficult to remove depending on the printer.  

Right now, one parameter changes both support and interface lines. It would be better if they were separated like in Cura, set the support line width to 80% and the interface to 30% of the nozzle diameter.