OrcaSlicer/OrcaSlicer #7683: Missing first layer infill on P1S in Orca at the edges

crs997, 2024-12-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

11

### Additional system information

5600x 6800xt

### Printer

Bambu Lab P1S

### How to reproduce

On my model with rounded edges, the infil has a gap where it starts. Compared to Bambu Studio there's a visible gap in where the infill starts. Using Arache solves the starting corner but not the finishing corner. I'll attach a screenshot of the two side by side so you can see what I mean.
[image]

### Actual results

This leaves tiny holes in the first layer

### Expected results

It should be filled in as it is in Bambu Studio. Using Classic Wall gen for both. 
[Rounded Edges Soap Dish 4 Orca Devs.zip](https://github.com/user-attachments/files/18048382/Rounded.Edges.Soap.Dish.4.Orca.Devs.zip)

### Project file & Debug log uploads

[Rounded Edges Soap Dish 4 Orca Devs.zip](https://github.com/user-attachments/files/18048397/Rounded.Edges.Soap.Dish.4.Orca.Devs.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

Pls don't make me have to go back to Bambu Studio.... sad face... I'm not sure why it does that, like i said above using Arache solves the starting corner but not the ending corner. that little squiggle that you can see in the Bambu studio file just means there's no gaps.

I tested using my Bambu A1 and it looks absolutely fine, it seems in my limited testing to be only the P1S,.

[image]

ThorsReign, 2024-12-10:
Have you tried setting your top/bottom infill wall overlap to 15%?  The top and bottom specific setting doesn't exist in bambu, it's for all walls but it's set to 15% and I see your Orca is set to 25% and I suspect that may be causing the difference, the overlap is resulting in less space for the line to fill that gap.

crs997, 2024-12-10:
You sir, are a legend. I had no idea that setting didn't exist in Bambu. So bambu has it set to 15% but it isn't an adjustable setting? 

That's bizarre but thank you very much for solving this issue for me. Because of the gap, sometimes the first layer would get caught. I haven't gotten home yet to fully confirm but re-slicing the model with it set to 15% has added that extra line in.

Crazy that Bambu don't tell you that yet in Orca it's set to 25% by default, crazy little situation but thanks so much!

closed 2025-03-19 by github-actions[bot]