OrcaSlicer/OrcaSlicer #7684: "Per-Element" Flow Ratio

tiziano149, 2024-12-07:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Not enough control over flow ratios, required for optimal results.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I have been 3D-Printing professionally for almost 15 years now and I have found that the ability to finely adjust flow ratios or extrusion multipliers is the key for sharp-looking prints. I have been doing this via gcode post-processing scripts for years, and this improved the quality of my prints significantly.

It would be a great improvement and even a USP over all other slicers if Orca did this from the slicer via a nice UI.

I have added screenshots of how this could look like in the UI.

[image]
[image]
[image]

I would love to see this implemented! :))

ShaneDelmore, 2024-12-08:
Can you share the script you use to adjust the flow ratios now?  I would like to test it and see if the results improve a variety of my prints.

tiziano149, 2024-12-09:
I used to do this back when I still used Simplify3D, it was a simple REPLACE function, as there are tags inside the GCODE (see image). I replaced the tag with another tag and two lines. One to add the extrusion multiplier and one for the K-Factor for pressure advanced. I declared a pair of values for eacht "Feature".

[image]

tiziano149, 2024-12-10:
The G-Code are:
Flow-Rate: **"M221 S100"**, where "S###" is the percentage.
Pressure Advance: **"M900 K0.1"**, where "K#" is the K-Factor

https://marlinfw.org/docs/gcode/M221.html
https://marlinfw.org/docs/gcode/M900.html

tiziano149, 2024-12-27:
[image]

Under Walls and Surfaces this function is already available for Top and Bottom surfaces.

It would so nice to have this feature for al the other "Elements" as well such as outer and inner walls, Infill, Gap Fill and so on...

tiziano149, 2025-04-12:
Okay, so there are at least three different issues regarding this same feature: #7684 (this one), #7768, and #8544. So there must be something to it. :D
What do we need to do to implement this? 🤔