OrcaSlicer/OrcaSlicer #7690: Data_dir for portable version

AsturTorQue, 2024-12-08:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I always have a portable version of Orcaslicer updated in my PC with Windows. 

And i had profiles saved of several filaments and settings.

The problem comes when I had to format my pc and I thought the portable version kept the profiles in the same directory as the portable version. Unfortunately formating PC I lost everything and later I realized this is kept in Appdata from C:\users\username folder...

This is extremely problematic when executing from USB portable device or if you have a versión kept in other partitions aparte from systemdrive.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Knowing there is a way executing the exe with the switch "--datadir" and pointing at the folder you like. I would suggest pointing by default to the same dirpath as the exe executed in portable version (maybe a file flag indicating it's a portable versión and creating the profile the same directory). And an option if you want to have it in Appdata or "locally"

marcinviz, 2024-12-11:
+1 for that. Last time i remember back when windows 95 was around, "portable" meant to be portable.

IntenseWiggling, 2025-01-26:
I had the same issue, but discovered that if you have a directory `data_dir` within the same directory as the executable (`orca-slicer.exe`), that is treated as as a data directory by default.

https://github.com/SoftFever/OrcaSlicer/blob/118e14d788b8c34700f5f6cef2bb66189329dca2/src/slic3r/GUI/GUI_App.cpp#L1922