OrcaSlicer/OrcaSlicer #7691: Keyboard Shortcuts For Prepare Window

jbowers01, 2024-12-08:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I'm a heavy keyboard shortcut user because of some neurological issues that sometimes make using a mouse or touchpad difficult. While navigating the prepare screen it's time-consuming and sometimes painful to make repetitive movements with my wrist or multiple fingers for touchpad gestures and having hotkeys would greatly improve my ability to use the slicer

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would like to have keyboard shortcuts to snap between multiple plates with direct combinations, such as ctrl-shift-1-9 or some combination of holding a key and tapping tab. Similar functionality for tabbing through objects on the plate instead of only being able to move them on the plate with keys would also be helpful, 

### Describe alternatives you've considered

An option to toggle behavior between moving an object or changing object/plate selection by holding a key and using directional keys to navigate would work just as well.

choovick, 2025-04-12:
yes please, i'm heavy hotkeys user as well... this is a bummer, at least ctrl-tab to or ctrl-shift-tab (osx) to navigate tab like an any other tab based app.

Felix14-v2, 2026-09-24:
@jbowers01 could you try the latest Nightly build? OrcaSlicer now has its own run bar (speed dial). You can quickly open it using space in any tab and jump to any printer/material/process setting, plate, layer or tab. Here's the demo:
https://youtu.be/DxJPhETR1Hs

Also now you can reassign almost all shortcuts and customize them to your needs. Please, give it a try! Feel free to suggest improvements and ping `@lamweilun`.

jbowers01, 2026-09-26:
This is great, I can see the benefit to this. However, when I had the idea I was thinking of something a little more simple like ALT+Tab to tab between plates or potentially ALT+plate number 01-36 to move directly to that plate. It might be asking a lot, but I think usability would be exponentially improved with specific plate switching shortcuts.

Felix14-v2, 2026-09-26:
> something a little more simple like ALT+Tab to tab between plates or potentially ALT+plate number

While it doesn't ship that out of the box, you can customize it and make something similar. The new dial has 10 slots for pinned actions: press Ctrl+B to pin an action, and then quickly run it using Alt+0-9. So you can combine that with the `Go to plate ...` action.