OrcaSlicer/OrcaSlicer #7694: Small area flow compensation not active in first layer

IRTrail, 2024-12-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

N/A

### Printer

Voron 2.4 / Voron Trident

### How to reproduce

Enable small area flow compensation.
Slice model.
Switch "Color Scheme" to "Flow"
Move the slider to the first layer and note no change in color.
Move the slider to the second layer and note color change.

First Layer:
[image]

Second Layer:
[image]

### Actual results

The change in flow is noticeable in the second layer, but not the first.
The typical overextrusion on small areas is apparent in the actual printing of parts, which causes artifacts in the subsequent few layers.

### Expected results

The first layer should have small area flow compensation active.

### Project file & Debug log uploads

[Electronics-Bay-Passthrough.zip](https://github.com/user-attachments/files/18052715/Electronics-Bay-Passthrough.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

kisslorand, 2024-12-11:
It's not a bug, it's intentional to avoid adhesion issues.

IRTrail, 2024-12-11:
Ok, but it would be nice if it worked on the first layer, possibly with an option to turn it off on the first layer if desired.

I guessed we'll call it a feature request then? 👍