OrcaSlicer/OrcaSlicer #7697: Assert error in lightModeColorFor() on start up

NiklausHofer, 2024-12-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Gentoo

### Additional system information

Architecture: x86_64; GCC: 14.2.1; CPU: i7-5930K; RAM: 24 GiB

### Printer

RatRig V-Core 4.1

### How to reproduce

1. Launch the software

### Actual results

A Popup appears, saying: "[...]src/slic3r/GUI/Widgets/StateColor.cpp(205): assert "iter != gLightColors.end()" failed in lightModeColorFor().". The window offers to show a back trace, too:
```C
wxEntry(int&, wchar_t**)
main
__libc_start_main
```

I can click "Continue". This then repeats for 3 times until the application stops for good. If I start in the terminal, here's what I get:
```
[2024-12-08 21:27:28.172070] [0x00007fd5af1ad700] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
/var/tmp/portage/media-gfx/orcaslicer-2.2.0/work/OrcaSlicer-2.2.0/src/slic3r/GUI/Widgets/StateColor.cpp(205): assert "iter != gLightColors.end()" failed in lightModeColorFor().
/var/tmp/portage/media-gfx/orcaslicer-2.2.0/work/OrcaSlicer-2.2.0/src/slic3r/GUI/Widgets/StateColor.cpp(205): assert "iter != gLightColors.end()" failed in lightModeColorFor().
/var/tmp/portage/media-gfx/orcaslicer-2.2.0/work/OrcaSlicer-2.2.0/src/slic3r/GUI/Widgets/StateColor.cpp(205): assert "iter != gLightColors.end()" failed in lightModeColorFor().
/var/tmp/portage/media-gfx/orcaslicer-2.2.0/work/OrcaSlicer-2.2.0/src/slic3r/GUI/Widgets/StateColor.cpp(205): assert "iter != gLightColors.end()" failed in lightModeColorFor().
Segmentation fault
```

To be clear, I never see the application main window, just that pop up with the error.

I tested on both x86_64 and arm64.

### Expected results

The application starts up.

### Project file & Debug log uploads

[2024-12-08 21:27:28.172070] [0x00007fd5af1ad700] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
/var/tmp/portage/media-gfx/orcaslicer-2.2.0/work/OrcaSlicer-2.2.0/src/slic3r/GUI/Widgets/StateColor.cpp(205): assert "iter != gLightColors.end()" failed in lightModeColorFor().
/var/tmp/portage/media-gfx/orcaslicer-2.2.0/work/OrcaSlicer-2.2.0/src/slic3r/GUI/Widgets/StateColor.cpp(205): assert "iter != gLightColors.end()" failed in lightModeColorFor().
/var/tmp/portage/media-gfx/orcaslicer-2.2.0/work/OrcaSlicer-2.2.0/src/slic3r/GUI/Widgets/StateColor.cpp(205): assert "iter != gLightColors.end()" failed in lightModeColorFor().
/var/tmp/portage/media-gfx/orcaslicer-2.2.0/work/OrcaSlicer-2.2.0/src/slic3r/GUI/Widgets/StateColor.cpp(205): assert "iter != gLightColors.end()" failed in lightModeColorFor().
Segmentation fault

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I built the application using this ebuild, here: https://github.com/thankjura/masterlay/blob/master/media-gfx/orcaslicer/orcaslicer-2.2.0.ebuild

ericghara, 2024-12-12:
Same thing happened with me.  I found this in a google search probably best to open a bug with the overlay maintainer.

NiklausHofer, 2024-12-12:
> Same thing happened with me, I found this while googling the problem. Probably best to open a bug with the overlay maintainer.

Figured as much: https://github.com/thankjura/masterlay/issues/5

closed 2025-03-21 by github-actions[bot]