OrcaSlicer/OrcaSlicer #7707: Overhangs Printed on top Surface with Solid Infil Overlap on

DerNinsaa, 2024-12-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

CPU: Ryzen 7 7800x3d, Display Card: Nvidia GeForce RTX 3090

### Printer

Voron 2,4

### How to reproduce

1. Provided STL: https://www.printables.com/model/570510-swiss-cheese-calibration-cube/files
2. Use Strength --> Top/bottom Shells --> Top/Bottom solid infill/wall overlap --> Value greater 5%
3. Slice Part

### Actual results

Sliced Part has Overhangs on top Surface, where there is no Layer/feature above. The Features are cut holes that clip the top surface and leave a small space, see Picture two

[image]
[image]

### Expected results

Expected Result shows when selecting Values lower than 5% for Top/Bottom solid infill/wall overlap.
Using any values largern than 5% results in the error shown above.
[image]

### Project file & Debug log uploads

[problematic_cube.zip](https://github.com/user-attachments/files/18061845/problematic_cube.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

DerNinsaa, 2024-12-09:
Out of curiosity, Printed the sliced Part, artefact is not visual only, but gets printed as shown.
Left is 25% Overlap, wich causes the issue, Right is 5% Overlap.
[image]

00raq00, 2024-12-17:
Bambu doesn't have this issue with the same profiles.

igiannakas, 2024-12-18:
can you try with one wall top surface turned off and report back please?

DerNinsaa, 2024-12-20:
> can you try with one wall top surface turned off and report back please?

Disabling One Wall on Top Surface fixes the weird slicing error.

DerNinsaa, 2025-04-13:
Problem solved with latest version.

closed 2025-04-13 by DerNinsaa