OrcaSlicer/OrcaSlicer #7715: Slicing not adding top surfaces on curved/slanted surfaces

fireblade2534, 2024-12-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Bambu Lab X1c

### How to reproduce

1. Load the project in orca slicer. 
2. Click the down arrow and click slice all
3. Go to the 1st plate
4. Turn on Line Type
5. On the bow there is no top surface line type

### Actual results

When slicing with Orca Slicer It does not add top layers on the bow even when using default settings
[image]

### Expected results

When slicing with Bambu Studio it does add top layers on the bow even when using default settings
[image]

### Project file & Debug log uploads

Orca Slicer project -> [Ship_Orca_Slicer.zip](https://github.com/user-attachments/files/18072041/Ship_Orca_Slicer.zip)
Bambu Studio project -> [Ship_Bambu_Studio.zip](https://github.com/user-attachments/files/18072046/Ship_Bambu_Studio.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

DerNinsaa, 2024-12-10:
This seems to be the same Issue as I documented https://github.com/SoftFever/OrcaSlicer/issues/7707
Somehow the Top Layer does not get generated over Overhangs.

Could you try reducing the setting I found that helps with this? Strength --> Top/bottom Shells --> Top/Bottom solid infill/wall overlap --> Value greater 5%

fireblade2534, 2024-12-10:
[image]
Nope still doesn't generate a roof

closed 2025-03-19 by github-actions[bot]