OrcaSlicer/OrcaSlicer #7718: [BUG] - no Use ASM for BambuLab A1 settings

DrumClock, 2024-12-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

BambuLab A1 + AMS lite

### How to reproduce

-

### Actual results

-

### Expected results

-

### Project file & Debug log uploads

Project + Gcode
[BAMBU A1.zip](https://github.com/user-attachments/files/18086049/BAMBU.A1.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

I wanted to print 4-colors on Bambulab A1 with AMS lite. Unfortunately, even though T0,T1,T2 and T3 are in the g-code,
the filament will not be replaced. The printer is not on WLAN but I print from an SD card.

"USE AMS" is not displayed 

Display
[image]

DrumClock, 2024-12-11:
I figured out that this option is needed:

'Export plate sliced ​​file' and save as 'xxx.gcode.3mf' on SD card

Again Bambulab (China) must have something extra and not standard G-code.
They simply will not comply with any standards.

closed 2024-12-11 by DrumClock

kekkodance, 2024-12-23:
the AMS doesn't conform to any standards simply because there's no standards that accomodate it, if BBL didn't want to comply to anything they could have made every G-Code operation different from other printers, instead they're similar

DrumClock, 2024-12-24:
Hi,
the point is that there are printers that have multiple heads or a material changer
and they don't need the entire 3mf project to print. They just need the standard g-code from T0,T1...

But not BBL, they just have to have something extra. It's nothing else.
It just doesn't support pure G-code with T0,T1,T2 ..etc.