OrcaSlicer/OrcaSlicer #7722: Filament change variable for a color

legotrain333, 2024-12-11:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The problem is there is no variable for color and filament changes

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

There to be a easy way to define color for filament changes so that Aidan macro can pick it up and display it on screen.

Like a way to have it so that during a filament change a message can pop up on your printer screen and say change to this color automatically based off of the filament colors in the slicer.

vgdh, 2024-12-11:
There is such a variable. I use it in my clipper macro.
FILAMENT_CHANGE FILAMENT_NAME='[filament_preset]'

legotrain333, 2024-12-12:
can I get your macro

vgdh, 2024-12-12:
```
[gcode_macro FILAMENT_CHANGE] # M600
gcode:
  {% set MM_TO_PURGE = params.MM_TO_PURGE|default(30)|int %}
  {% set FILAMENT_NAME = params.FILAMENT_NAME|default('The name of the filament has been not passed')|string %}
  
  RESPOND TYPE=echo MSG='FILAMENT CHANGE (M600) command. Purge {MM_TO_PURGE} mm'
  
  PAUSE
  #SET_GCODE_VARIABLE MACRO=global_vars VARIABLE=caller VALUE='"M600"'
  SET_GCODE_VARIABLE MACRO=global_vars VARIABLE=m600_purge_mm VALUE={MM_TO_PURGE}
  
  PLAY_SHORT_SONG
  UNLOAD_FILAMENT
  
  _CHANGE_FILAMENT_DIALOG FILAMENT_NAME='{FILAMENT_NAME}'
  

[gcode_macro _CHANGE_FILAMENT_DIALOG]
gcode:
  {% set m600_purge_mm = printer["gcode_macro global_vars"].m600_purge_mm %} 
  {% set FILAMENT_NAME = params.FILAMENT_NAME|default('The name of the filament has been not passed')|string %}
  RESPOND TYPE=echo MSG='{FILAMENT_NAME}'
  
  RESPOND TYPE=command MSG="action:prompt_begin M600 color changing"
  RESPOND TYPE=command MSG="action:prompt_text Insert the new FILAMENT and press LOAD then CONTINUE"
  RESPOND TYPE=command MSG="action:prompt_text {FILAMENT_NAME}"
  RESPOND TYPE=command MSG="action:prompt_button LOAD FILAMENT|LOAD_FILAMENT MM_TO_PURGE={m600_purge_mm}|primary"
  RESPOND TYPE=command MSG="action:prompt_button CLOSE THIS DIALOG|RESPOND TYPE=command MSG=action:prompt_end|warning"
  RESPOND TYPE=command MSG="action:prompt_button CONTINUE PRINT|_resume_and_close_dialog|secondary"
  RESPOND TYPE=command MSG="action:prompt_show"

[gcode_macro _resume_and_close_dialog]
gcode:
  RESPOND type="command" msg="action:prompt_end"
  RESUME
  

[gcode_macro UNLOAD_FILAMENT] 
gcode:
  RESPOND TYPE=echo MSG='PURGE Unload filament'
  M83 ; set extruder to relative

  # jerk the filament out of the melt zone
  {% for i in range(0, 5) %}
    G1 E-10.5 F3000 
    G4 P300
    G1 E10 F3000  
    G4 P300
  {% endfor %}

  G1 E-20 F3000  
  G4 P2000 # wait ms. Crystalize filament
  G1 E-20 F3000 # wait ms. Crystalize filament
  G4 P2000 # wait ms. Crystalize filament
  G1 E-40 F3000                 ; retract the rest of the way
  M82                            ; set extruder to absolute  

[gcode_macro LOAD_FILAMENT]
gcode:
  {% set MM_TO_PURGE = params.MM_TO_PURGE|default(31)|int %}
  {% set PURGE_STEP_MM = 5|int %}
  {% set PURGE_SPEED = 300|int %}

  RESPOND TYPE=echo MSG='Loading filament and PURGE {MM_TO_PURGE} mm'
  
  M83  ; set extruder to relative
  # 70mm from gears to the nozzle (kp3s)
  G1 E50 F1500  ; load
  G1 E25 F{PURGE_SPEED} # load slow

  {% for i in range(0, 1000) %}
    {% if MM_TO_PURGE > PURGE_STEP_MM * i %}
      {% set left_to_purge = MM_TO_PURGE - PURGE_STEP_MM * i %}
      
      {% if left_to_purge > PURGE_STEP_MM %}
        RESPOND TYPE=echo MSG='PURGE {PURGE_STEP_MM}mm'
        G1 E{PURGE_STEP_MM} F{PURGE_SPEED}    
      {% else %}  
        RESPOND TYPE=echo MSG='PURGE {left_to_purge}mm'
        G1 E{left_to_purge} F{PURGE_SPEED}    
      {% endif %} 
      G4 P300 # wait ms
    {% endif %} 
  {% endfor %}
 
  M82                            ; set extruder to absolute  
  #G4 P5000 # wait ms
  RESPOND TYPE=echo MSG='Load filament is finished'
  
```

legotrain333, 2024-12-13:
thank you. How do I get the name of the fliment in the box that opens.

vgdh, 2024-12-14:
I'am not sure I've understood you correctly. I use name convention for the filaments like: `kingroon petg RED` etc. 
So this name of the filament will appear in the dialogue box in the klipper.

vgdh, 2024-12-14:
> thank you. How do I get the name of the filament in the box that opens. Is there also a way for when you color the models with the different filament colors in the filament tab in OrcaSlicer for to automatically populate the dialogue box that's opens up in Klipper with the corresponding color.

I'am not sure I've understood you correctly. I use name convention for the filaments like: kingroon petg RED etc.
So this name of the filament will appear in the dialogue box in the klipper.

legotrain333, 2024-12-14:
That makes a lot of sense and. In the filament change machine code in OrcaSlicer what do I have to put.

vgdh, 2024-12-15:
I use this code in `change filament G-code` section 
```
M104 S[new_filament_temp]   ; set temperature before filament change
FILAMENT_CHANGE FILAMENT_NAME='[filament_preset]' 
```