OrcaSlicer/OrcaSlicer #7724: Infill Advances settings drop down boxes not resizing

living-relation, 2024-12-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11 Pro 24H2 26100.2314

### Additional system information

Intel Core i9-10850K CPU @ 4.50GHz, 32GB RAM, Nvidia Quadro RTX 4000

### Printer

K1

### How to reproduce

Process [select advanced settings] -> Strength -> Infill. 
Sparse infill anchor length and or Maximum length of the infill anchor. 

[image]

### Actual results

choosing  [Sparse infill anchor length] and/or [Max length of the infill anchor] from either drop down causes the text to be completely hidden so the state of your chosen setting is not viewable. no matter what the scaling of the window is or the size of the panel in Orcaslicer.

### Expected results

when an option is chosen from either drop down menu, resize it to fit the text, or set the box to be a fixed size large enough to show whichever option is the one with the longest string. that way, all of them would always be visible.

### Project file & Debug log uploads

[klippy (6).log](https://github.com/user-attachments/files/18103570/klippy.6.log)
[backup-fluidd-v1.31.2-K1.json](https://github.com/user-attachments/files/18103572/backup-fluidd-v1.31.2-K1.json)
[drive-download-20241211T225120Z-001.zip](https://github.com/user-attachments/files/18103578/drive-download-20241211T225120Z-001.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]

discip, 2024-12-12:
@living-relation
Please try the latest dev build:
https://github.com/SoftFever/OrcaSlicer/actions/runs/12170662660/artifacts/2276781512

living-relation, 2024-12-14:
> @living-relation Please try the latest dev build: https://github.com/SoftFever/OrcaSlicer/actions/runs/12170662660/artifacts/2276781512

Thanks. That mostly fixed it. It still cuts off a decent portion of some of the options, though.

closed 2025-03-23 by github-actions[bot]