OrcaSlicer/OrcaSlicer #7726: Failed to slice with error "vector" or crash.

fenhelix, 2024-12-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

macOS 15.1.1 (24B91)

### Additional system information

Mac Studio 2023 Apple M2 Ultra 24 cores (16 performance and 8 efficiency)
64 GB RAM

### Printer

Bambu Lab X1 Carbon 0.6 nozzle

### How to reproduce

Open file. 
Click Preview.

Note:
This happens with 3 walls but does not happen with 2 or 4 walls.
It also does not happen with Arachne.

### Actual results

An error appears that says one of two things depending on infill selected.
[image]
[image]

May also crash (report below)

### Expected results

The file should be sliced. 

### Project file & Debug log uploads

[debug_Wed_Dec_11_19_12_27_73891.zip](https://github.com/user-attachments/files/18104230/debug_Wed_Dec_11_19_12_27_73891.zip)
 
[Log.zip](https://github.com/user-attachments/files/18104333/Log.zip)

[SignMountBody.3mf.zip](https://github.com/user-attachments/files/18104339/SignMountBody.3mf.zip)
[Crash.log](https://github.com/user-attachments/files/18104341/Crash.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

jb510, 2024-12-21:
Perhaps helpful for debugging.  I ran into the same "vector" issue slicing a gridfinity base plate, 
GridLite 7x7.stl  
https://www.printables.com/model/413761-gridfinity-refined/files
I was slicing with my default 3 walls.  Per this issue, I tried changing it to 4 and it sliced successfully.  
[image]

Also notable, when I had added this STL to a new plate in another project and tried to slice it it crashed Orca entirely (unexpected exit/quit).  Putting this file in it's own new project results in the "vector" message.

closed 2025-05-22 by github-actions[bot]

bpengu1n, 2025-05-29:
I am getting this error as well; it happens with either 2 or 3 walls, but does not happen when I set to 4 walls. 

For me, I will either get the error "vector"; or OrcaSlicer will just crash altogether if I attempt to slice again after receiving the error (whether or not I have sliced again successfully - for example, if I receive the error, raise walls to 4 and successfully slice; and then reduce back to 2 and attempt to slice again, OrcaSlicer will crash without giving any error).

Attaching the offending 3mf in ZIP.

[error - vector.3mf.zip](https://github.com/user-attachments/files/20496358/error.-.vector.3mf.zip)

bpengu1n, 2025-05-29:
Happy to provide any other debugging info if I know where to grab it! I would debug myself but don't have time at the moment to set up a dev environment for Orca...