OrcaSlicer/OrcaSlicer #7735: Filament Fan Speed for Overhangs Ignored Unless Slow Down for Overhangs Enabled

chris99336, 2024-12-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu X1C

### How to reproduce

Put in an overhang test part. Under filament settings, choose force cooling for overhangs and bridges. Set an overhang threshold and fan speed. Slice print.

### Actual results

Threshold and fan speed make absolutely no difference to the fan in the sliced print. However, if you choose slow down for overhangs in the print settings, and set all the overhangs that you don't want to slow down for to 99% and the ones that you do want to slow down for to whatever value you want, then suddenly the overhangs are detected and the fan speed is respected.

### Expected results

The fan speeds should be respected regardless of slow down for overhangs. Also, while I am going to mention it here, I am going to submit a feature request to have more than one threshold for overhang fan speed.

### Project file & Debug log uploads

[log for bug report.zip](https://github.com/user-attachments/files/18119705/log.for.bug.report.zip)
[overhang_numbers for bug report.zip](https://github.com/user-attachments/files/18119712/overhang_numbers.for.bug.report.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Boerny41, 2025-01-12:
Can confirm this behavior. Also, enabling slow down for overhang but setting all the speed to the same, but not the outer/inner wall value, results in just a few of the overhangs getting recognized. To get them all recognized again, all overhang speeds must be different from each other and the outer/inner wall speed.

chris99336, 2025-05-10:
> Can confirm this behavior. Also, enabling slow down for overhang but setting all the speed to the same, but not the outer/inner wall value, results in just a few of the overhangs getting recognized. [...]

I hadn't realized that. Thanks for the input. Maybe it wasn't an issue for mine cuz I set mine all to 99% instead of explicit speeds.

chris99336, 2025-05-10:
> Adding a comment to try to get this more attention. I noticed this as well just now. "Slow down for overhangs" causes artifacts when printing high temp filament, but with this bug I have to choose between those artifacts and low quality overhangs due to lack of fan speed.

I too hope this gets more attention though I haven't considered it too much more since finding that workaround but I'm sure it is still causing some quality reductions.

kolyan4g, 2025-08-24:
Here's another example. But then I found out that it was enough to slow down by 2 mm/s for all overhang options (relative to 40 mm/s). I don't know if the actual value or the percentage is important here. With the percentage, the fan only started to accelerate at 96% speed on overhangs.

[image]

[image]

[image]

[image]