OrcaSlicer/OrcaSlicer #7740: Volume flow rate and nozzle diameter

drcowsley, 2024-12-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

There is no simple relationship between temperature, nozzle diameter and flow rate.  It depends on the plastic.  Right now finding what works is hit and miss.
Some filament suppliers are addressing the issue but as far as I can see, in an uncoordinated way not, always addressing the root problem of non-Newtonian flows where the relationship between  pressure and flow rate is non-linear.  
This is essentially a filament issue.
 If Orca could encourage the use of a simple table of temperature (range), nozzle diameter and volume flow rate limit for a particular filament rather than the current single figure flow rate limit, it might help the technology to get to the best solution sooner.
Filament suppliers don't seem to be coming to a coherent solution.  Can Orca step up and lead the way?

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

When I need a smaller nozzle for more precision, or a larger one for speed, or possibly even two different size nozzles for different features of the same model, I want to have a fighting chance of getting somewhere near on the first prototype.

### Describe alternatives you've considered

Including a table in the single material record is the right solution.  Different records for different nozzle sizes is an option but risks data duplication.
Flow rate is not the only feature that depends unpredictably on nozzle cross section area.   Stringing does too, so retraction parameters might also belong in the table (or a table).

VigilantGardener, 2024-12-14:
Help me understand this please. 

Is it not standard practice to create a new filament profile and calibrate it when setting up to use a different size nozzle with a given filament?

drcowsley, 2024-12-15:
Hello and thank you for responding.
No - it is not my standard practice.  I use the information provided by the
manufacturer and fine tune nozzle temperature and retraction length to deal
with level bonding and stringing.  I am a hobby user, not a professional.
Using published data makes sense to me, especially because getting the feed
rate too fast or in the case of PLA in Summer, too slow can cause feed jams
and a strip down of the hot end.  That is even with the starting point the
manufacturers give me.  If I am trying to make a precision part in PETG
because it has to be tough in service it's a nightmare with a 0.2mm nozzle
and can take forever if I get it too wrong.
I suspect the filament makers know all this, and that is why they publish
as much information as they do, and in the way they do, but none of the
suppliers I can afford to use do much more than print a temperature range
on the spool.  For their high speed filament Sunlu do give temperature
ranges for different print speeds assuming a 0.4mm nozzle I assume, and
Orca rather more sensibly uses a maximum feed rate parameter, again for a
0.4mm nozzle and 'their own' PLA, but what would that figure be for a 0.2
nozzle?  It is unlikely to be pro-rata to area.  For one person to find a
good starting point and publish it, as happens for 0.4mm, would make more
sense than everybody doing it.

Does this make the issue clearer?

DrC

VigilantGardener, 2024-12-16:
What you're describing has multiple facets outside of simply the slicer software.  Basically it's the ecosystem of Bambu products that you just described.  Unless you are going to buy into that ecosystem and rely on it, then you'll have to start tuning your own printer, filament, and print profiles.  There is no escaping this reality.

This guide will provide you a good starting point to learn how to perform the necessary calibrations and tests, all of which can be performed before attempting to produce a final product.
https://teachingtechyt.github.io/calibration.html?fbclid=IwAR0ZnBA6qRtcqT-L3FrW-8704U7IAObW1yk3CmGjyHPcRFMCyULCjDhBzPA#intro

drcowsley, 2024-12-17:
Thank you for that reference.  I have limited resource for in depth study,
testing and interpreting, since my task purpose is one-offs to please
friends. Prototype issues are my interest, not process issues.
I do instinctively buy into constructive ecosystems, and in that context my
suggestion was intended as a way of making the slicer environment a better
fit with the larger eco-system - at least for me.

I will be following your link.  Thank you again.

DrC