OrcaSlicer/OrcaSlicer #7741: Measuring diameters

asp1972, 2024-12-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Bambu X1C

### How to reproduce

1. Сreate a primitive cylinder.
2. Measure the diameter.
3. Get an unclear result
[image]

### Actual results

Incorrect diameter measurement

### Expected results

Сorrect diameter measurement.

p.s. In Bambustudio where did you get the measurement block, the diameter size is shown correctly
[image]

### Project file & Debug log uploads

[Cylinder.zip](https://github.com/user-attachments/files/18135573/Cylinder.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

limited660, 2025-02-06:
Also seeing this on Windows 11, 2.3.0-dev build 658c4f0.

kvikindi, 2025-02-11:
Similar issue seen here on linux mint debian, flatpak release 4d762c4, as well as Win 10 portable release f11eb34

PR #8575 (2025-02-26): Measure: Fix wrong diameter when selecting a circle feature
Fix #7741
closed 2025-02-26 by Noisyfox