OrcaSlicer/OrcaSlicer #7742: Appimage doesn't start due missing  libwebkit2gtk-4.0.so.37

CaptainCoward, 2024-12-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0  and 2.3.0 dev

### Operating System (OS)

Linux

### OS Version

EndeavourOS

### Printer

Bambu Lab X1

### How to reproduce

Not much to do, just try to run the Appimage 

### Actual results

Appimage doesn't start due missing component

### Expected results

Should start

### Project file & Debug log uploads

/tmp/.mount_OrcaSlJbRwU9/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MxBrnr, 2024-12-14:
Looks like you did NOT actually search for existing issues, because this is a duplicate of https://github.com/SoftFever/OrcaSlicer/issues/4616

As it currently stands, there is unlikely to be a fix for this because Orca is in the process of moving to Flatpak instead of AppImage, the same as PrusaSlicer has now done and will be exclusive with for all future releases.

CaptainCoward, 2024-12-15:
Ty i must have been stupid. I searched for solutions and copy and pasted so often that i must have done something wrong when looking it up. Should have checked again. I'll close the issue.

closed 2024-12-15 by CaptainCoward