OrcaSlicer/OrcaSlicer #7743: Attempt to set bed tempurature to 115°C only yeld temperature of 110°C

SaltyCodger-YT, 2024-12-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

AMD 3950X 128GB RAM, 4070 RTX, 20TB HDD space split between different drives

### Printer

Creality K1 Max 

### How to reproduce

1)  Go to Filament presets 
2)  Set current bed type to 115°C
3)  Print a part
4) Observe bed temperature in Fluidd

### Actual results

Fluid shows a temperature set at only 110°C, The Temperature will max out at 110°C. You can enter 115°C in Fluid and the temperature will rise to the 115°C Temperature until the next temperature G code.

### Expected results

With a setting of 115°C, I would expect the build plate to attempt to reach 115°C. Especially if I can make the change manually in Fluidd.

### Project file & Debug log uploads

[Output for Report.zip](https://github.com/user-attachments/files/18136160/Output.for.Report.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

vgdh, 2024-12-14:
please provide gcode file

tlhintoq, 2024-12-15:
115° is outside the range of supported materials of PLA/PETG/TPU/ABS.
Even ABS is recommended at 110°.
Maybe Orca knows better and doesn't want to let you kill your printer???
[image]

https://www.matterhackers.com/articles/how-to-succeed-when-printing-with-abs
[image]

SaltyCodger-YT, 2024-12-15:
I am printing ABS getting significant warping and experimenting to locate the cause. Fans are at < 10%, if 115 worked then I need to invest in a chamber heater.
AJF

 
 

image.png (view on web)

https://www.matterhackers.com/articles/how-to-succeed-when-printing-with-abs
image.png (view on web)

yoyo42, 2024-12-15:
If you look under config at the printer.cfg file for your printer and find the [heater_bed] section what is the max_temp there set to? If that's 110 then that's as high as Orca thinks our printer is able to go.

This is mine for a QIDI Plus4:

**[heater_bed]**
heater_pin: U_1:PB10
sensor_type:NTC 100K MGB18-104F39050L32
sensor_pin:U_1: PA0
max_power: 1.0
control = pid
pid_Kp=63.418 
pid_Ki=1.342 
pid_Kd=749.125
pwm_cycle_time:0.001
min_temp: -60
**max_temp: 125**

tlhintoq, 2024-12-16:
If you are REALLY insistent on changing it to 115 you could just change it in the gcode before you send it.
If the printer accepts it, cool.  If it says "oh heck no" and limits that request to the firmware max then you know that too.

trampas, 2025-03-20:
I am using 2.3.0 and ran into the same bug.  I am printing polycarbonate and want bed temperature higher than 110. 

The printer.cfg has max bed at 150C. 

In the filament settings I changed to 1`20C and orca slicer only outputs 110C max.   This appears to be hard coded somewhere in orca slicer from what I can find.

closed 2025-06-27 by github-actions[bot]