OrcaSlicer/OrcaSlicer #7745: Arrangement with Print by object

3dfanattic, 2024-12-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Bambu Lab A1

### How to reproduce

1. Set print sequence to by object
2. Have a plate with a group of objects.
3. Have two other groups of objects off the plate, not on another plate.
4. Arrange

[image]

### Actual results

Objects should arrange according to the set print sequence.
(I know this is easily remedied by just arranging again but thought I'd submit it.)
[image]

### Expected results

[image]

### Project file & Debug log uploads

[bug report plated-accessories x1.zip](https://github.com/user-attachments/files/18137582/bug.report.plated-accessories.x1.zip)
[log.zip](https://github.com/user-attachments/files/18137585/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tlhintoq, 2024-12-16:
If someone grabs this ticket for looking at doing this `by print sequence` it might be nice to address this other issue that's probably closely related and get a significant improvement all at once.
https://github.com/SoftFever/OrcaSlicer/issues/5153

Noisyfox, 2025-01-14:
There seems to be a bug that the clearance box (the line around the object) is mistakingly too small in certain cases, which led to this problem. Will take a look. I believe https://github.com/SoftFever/OrcaSlicer/issues/5153 is caused by the same problem and will take a look too.

Noisyfox, 2025-01-14:
The issues are introduced here https://github.com/SoftFever/OrcaSlicer/commit/0b6a1d36360448c5e6818001b9d8ea99a57567a7
due to the fact that the new `Print::is_all_objects_are_short()` function does not count in any object that is off plate, which breaks the arragement as well as the fill instance with bed.

@vovodroid could you please take a look that this behavior is desired for the per-skirt feature (so that there will be two versions of this function that one ignores off-plate objs and one does not)?

PR #8038 (2025-01-22): Fix arrange when print-by-object and have object off-plate
Fix #5153
Fix #7745

Partly revert 0b6a1d36360448c5e6818001b9d8ea99a57567a7
closed 2025-01-22 by Noisyfox